commaai/openpilot #30518: 2017 Toyota Prius w/ 3x & SDSU

nicholas-a-stoner, 2023-11-24:
### Describe the bug

Upon startup Openpilot will throw a Cruise Fault error with a prompt to restart the car. Restarting the car does not clear the error. I've found that _sometimes_ unplugging the USB-C cable on the 3x and then plugging it in again will clear the error. 

### Which car does this affect?

Toyota Prius 2017

### Provide a route where the issue occurs

46c5fbcff27c4973|2023-11-22--07-01-44--0

### openpilot version

0.9.5

sshane, 2023-11-29:
It looks to me that something is wrong with your SDSU. PRE_COLLISION did not show up for 10 seconds, as well as ACC_CONTROL, DSU_CRUISE, etc.

What is the source of this SDSU?

[image]

nicholas-a-stoner, 2023-11-29:
DSU came from eBay and then was sent with a black Panda to Erich for
modification. He did say that using a black Panda was something he hadn't
done before.

ErichMoraga, 2023-11-30:
The normal way to handle issue for SDSUs purchased from me is to contact me, not open an OP GHI.  In this situation, I shipped Discord user `@starvinmarvin0235 / @starvinmarvin420` his SDSU 2 months ago...
https://tools.usps.com/go/TrackConfirmAction?tLabels=9400111206215032582071
...delivered September 29th.  It tested perfectly in my 2017 Prius Prime prior to shipping, and surely worked fine upon receipt.  

@nicholas-a-stoner  When did it stop working, and why didn't you DM me?

sshane, 2023-11-30:
If this is not an issue with the SDSU/DSU, feel free to comment back.

closed 2023-11-30 by sshane