commaai/openpilot #30527: Ford maverick 2022 master CI is broken

ayeganov, 2023-11-26:
### Describe the bug

With the update from the last few days I am getting a report of radar malfunction. Once I disconnect OP the error goes away. 

### Which car does this affect?

Ford maverick 2022 

### Provide a route where the issue occurs

Gjhgjh

### openpilot version

0.9.6

ayeganov, 2023-11-27:
@adeebshihadeh My dongle id is [c251cea6f2c6d154](https://useradmin.comma.ai/?onebox=c251cea6f2c6d154)

adeebshihadeh, 2023-11-27:
> @adeebshihadeh My dongle id is [c251cea6f2c6d154](https://useradmin.comma.ai/?onebox=c251cea6f2c6d154)

Can you upload the full logs from a route with the issue? I only need the first segment.

adeebshihadeh, 2023-11-28:
Fixed on master, and should be on master-ci soon too.

closed 2023-11-28 by adeebshihadeh