commaai/openpilot #30530: 2020 Ford Escape

Bolender1983, 2023-11-26:
### Describe the bug

Dashcam mode car unrecognized error

### Provide a route where the issue occurs

561745cc72a08831

### openpilot version

0.9.6

Bolender1983, 2023-11-26:
0.9.5 is the version

sshane, 2023-11-28:
Can you try this branch? https://github.com/commaai/openpilot/tree/escape-2020-fw

Installer URL:

https://installer.comma.ai/commaai//escape-2020-fw

Bolender1983, 2023-11-29:
When I put this url in the comma started working but now it will not let
collusion asst stay on and diables my front view camera.

Thank you for your help.

sshane, 2023-11-29:
The AEB disabled icon is shown in your cluster if you turn on openpilot longitudinal at this time. Can you try without that and see if everything works? Also, openpilot should still work in that state.

sshane, 2023-11-29:
Oh I see the blocked msgs, hmm everything is getting blocked despite it passing offline tests on your route. Have you made any changes to panda safety? Try to re-flash it with `./panda/board/flash.py`

Bolender1983, 2023-11-29:
I have not try to refresh the panda, I am not for sure how to do that

sshane, 2023-11-29:
Try to factory reset and re-install that branch, then try without openpilot longitudinal (alpha).

Bolender1983, 2023-11-29:
Ok, so I've reinstalled it, but this time it is now showing pilot
unavailable. Adapted cruise control disabled and it will not let my cruise
control to set

sshane, 2023-11-29:
Okay great, no blocked messages now. I think you enabled normal "dumb" cruise control by accident. Try to reverse the actions of this video. Can you take a photo of that cruise control setting? https://www.youtube.com/watch?v=6c0SDhA1GP0

sshane, 2023-12-07:
@Bolender1983 does your vehicle come with adaptive cruise control?

drfrink, 2023-12-29:
I'm running into the same issue with 2022 ford escape, comma is in dashcam mode

sshane, 2024-01-03:
Closing since you returned your device @Bolender1983.

sshane, 2024-01-03:
@drfrink please make a new issue if you haven't already.

closed 2024-01-03 by sshane