commaai/openpilot #30532: Car not recognized after starting with remote start

brugsy, 2023-11-27:
### Describe the bug

Observed when starting with remote start, car is not recognized and openpilot is in Dashcam only mode. Restarting vehicle will usually fix this issue.

There are several similar bugs about starting in dash cam mode, but this seems to be linked to remote start for my vehicle as a way it boots into dashcam mode.

### Which car does this affect?

Chevrolet Volt 2017

### Provide a route where the issue occurs

70cb85c2d0346410|2023-11-25--16-22-37

### openpilot version

fa9680911c3678beae00664b7eebd219b4778660

sshane, 2023-11-28:
Do you mean you turn the car on with remote start, comma device stays off, then you get in and fully turn the ignition on and it starts to dashcam mode?

sshane, 2023-12-07:
@brugsy is this the case?

sshane, 2024-01-03:
Any updates @brugsy?

sshane, 2024-01-03:
Closing since this hasn't happened for a few weeks and there's no rlogs to debug this. If this happens again, please report back here!

closed 2024-01-03 by sshane