commaai/openpilot #30565: 2023 Ford Explorer ST

David-prince1, 2023-11-30:
### Describe the bug

Comma 3 is not recognizing my 2023 Explorer ST. Is there another custom URL I should be using? 

### Which car does this affect?

2023 Ford Explorer ST

### Provide a route where the issue occurs

google_116479420611226073097

### openpilot version

0.8.10

sheaduncan, 2023-12-02:
That's a really old OP version. Have you tried just running installer.comma.ai/commaai/master ?

sheaduncan, 2023-12-03:
> > That's a really old OP version. Have you tried just running installer.comma.ai/commaai/master ?
> 
> I misspoke. I'm running version 0.9.5
> 
> I just tried installer.comma.ai/commaai/master with the same result. Still showing car unrecognized.

Can you do the following:

Go to: useradmin.comma.ai

click on route
click on the first segment
in the drop-down in the left corner, select "CarParams" and there are all your FW versions.

Can you reply with your FW Versions?

If you're interested in learning you can go to here and see if all your FW Versions are listed:
https://github.com/commaai/openpilot/blob/a2b48efa2083c86638b536df180fbb638cab7404/selfdrive/car/ford/values.py#L181

closed 2023-12-07 by sshane

RAINNY4114, 2024-02-28:
my Lincoln MKX carFW is:
 'carFw': [{'address': 1840,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'eps',
                          'fwVersion': b'K2GC-14D003-AH\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1848,
                          'subAddress': 0},
                         {'address': 1842,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'shiftByWire',
                          'fwVersion': b'HA1P-14C561-AA\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1850,
                          'subAddress': 0},
                         {'address': 1798,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'KA1T-14F397-AF\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1806,
                          'subAddress': 0},
                         {'address': 1888,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'abs',
                          'fwVersion': b'KG9C-2D053-MD\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1896,
                          'subAddress': 0},
                         {'address': 1892,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'JX7T-14D049-AD\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1900,
                          'subAddress': 0},
                         {'address': 1888,
                          'brand': 'mazda',
                          'bus': 0,
                          'ecu': 'abs',
                          'fwVersion': b'KG9C-2D053-MD\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x88'],
                          'responseAddress': 1896,
                          'subAddress': 0},
                         {'address': 1840,
                          'brand': 'mazda',
                          'bus': 0,
                          'ecu': 'eps',
                          'fwVersion': b'K2GC-14D003-AH\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x88'],
                          'responseAddress': 1848,
                          'subAddress': 0},
                         {'address': 1798,
                          'brand': 'mazda',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'KA1T-14F397-AF\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x88'],
                          'responseAddress': 1806,
                          'subAddress': 0},
                         {'address': 1892,
                          'brand': 'mazda',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'JX7T-14D049-AD\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x88'],
                          'responseAddress': 1900,
                          'subAddress': 0}],
               'carName': 'ford',
               'carVin': '2LMPJ8L96JBL37843',