commaai/openpilot #30571: LKAS Fault

Luft1950, 2023-12-02:
### Describe the bug

After the release of OpenPilot 0.9.5 I have been getting daily LKAS Faults on my 2023 Bolt EUV.

### Which car does this affect?

2023 Bolt EUV

### Provide a route where the issue occurs

I-5 Olympia to Highway 512 East and then to Puyallup.

### openpilot version

0.9.5

### Additional info

0.9.4 worked okay but it seems to occur for no reason on 0.9.5. Very unchill.

Luft1950, 2023-12-02:
> What's your dongle ID? Do you have a route where this happens?

Hi Adeeb, 

My dongle id is c93a84c2e28979fe. I travel from Olympia WA to the 512 cutoff then to Puyallup WA. 

Thanks,.

adeebshihadeh, 2023-12-02:
Thanks, I'm looking into it.

sshane, 2023-12-07:
In one of those cases, right before the LKAS fault, I saw that an LKAS message/counter was missing on the 128 read back bus, so openpilot correctly re-sent the same counter, then it immediately faulted. Usually the EPS is robust to sending at a slower rate, it looked like the message might have actually got sent, but not added to bus 128? `pandaStates` indicates no dropped messages though.

[image]

Luft1950, 2023-12-07:
Thanks Shane. I can't say that I understand a lot about what you said but it's good to know that you're on it. Any chance this will be fixed in the next version of OpenPilot? 
Thanks again.

nharada1, 2024-01-31:
For what it's worth I see the same issue on my 2022 Bolt EUV

siddchou, 2024-02-01:
I am also getting similar issue in Mazda 2024 every time during my 1st trip each day . 

e548ddc194075bfb|2024-01-31--16-12-06--0

sshane, 2024-03-20:
@Luft1950 does this still happen?

Luft1950, 2024-03-20:
I can let you know in about two weeks. It only happened when I drove from Tenino to Puyallup for my chemo treatments. I'll be starting my next cycle this coming Friday and will last till Thursday.

Luft1950, 2024-04-08:
> Shane, This last week it did happen once. Tuesday March 26th on the first drive. (That video is about to be rotated off.) It was weird. Normally I get three "Take Control Immediately before it throws a fault but this time I think it only game me one.

Shane, the LKAS fault happened again just today. No "Take control" warnings at all. The video is for Monday April 8th 12:50 to 13:38.

nharada1, 2024-04-08:
I ended up simply returning my Comma, this issue was too prevalent to safely drive.

sshane, 2024-04-08:
@Luft1950 I requested upload, please put the device on WiFi for 30 minutes to let the logs upload.

Luft1950, 2024-04-09:
Just got your request. I hope you were able to get the upload. The car is on WiFi now.

sshane, 2024-04-09:
Looks like there is a problem with the panda reporting that the message was delivered. openpilot then re-sends the same counter as it thinks it wasn't delivered, but since it faulted here and there's no blocked or lost tx messages supposedly it did actually get sent.

[image]

Luft1950, 2024-04-28:
Shane, I don't know if you want me to tell you about each LKAS fault. If not just let me know. 
Today I got two. One when I was on my way to the cancer center and one when I went home. The videos are Sat. Apr 27 11:46 to 12:45 and Sat Apr 27 13:18 to 14:05. 
Thanks.

Luft1950, 2024-04-28:
Shane, I got another LKAS fault today. The video is Sunday, April 28, 7:16am to 8:12 am. 

Thanks.

Luft1950, 2024-04-30:
Shane, I got another LKAS fault yesterday. It was a weird one. I was returning from my chemo and had received two "Take control immediately" warnings so I disengaged OpenPilot before the fault. I then drove without OpenPilot to Olympia. I looked up and the Comma 3X was displaying the LKAS fault! So my device faulted while being disengaged. The video is Monday April 29, 15:19 to 16:36. 
Thanks

sshane, 2024-04-30:
Can you go in connect and send the link for each?

sshane, 2024-05-01:
@Luft1950 You're the only one hitting this [old issue](https://github.com/commaai/openpilot/issues/26016). Currently this appears to be a software bug, but we don't currently have the time to debug the root cause for your specific case. We are going to allow you to return your device if you would like, just send this GitHub thread: https://shop.comma.ai/a/returns

Related to https://github.com/commaai/openpilot/issues/28899, the CAN packet dropped out for a few frames where we were expecting the read back bus to say that the message was transmitted, so we transmitted it again incorrectly.

[image]

closed 2024-05-02 by adeebshihadeh