commaai/openpilot #30598: 2017 Toyota Highlander "paramsd temporary error"

jtexp, 2023-12-04:
### Describe the bug

When enabling the Toyota's ACC, within milliseconds the comma3x with openpilot release and master bd0ab95 will give UI error "params temporary error"  Then the Toyota driver screen will say Radar Cruise unavailable. The radar cruise works fine with the 3x unplugged and the ACC will say "Radar Ready" before engagement.  

### Which car does this affect?

Toyota Highlander 2017

### Provide a route where the issue occurs

43a990eec4811111|2023-12-04--11-06-56--0

### openpilot version

Issue is on release and bd0ab95

adeebshihadeh, 2023-12-04:
Looks like your steering alignment may be off. 

@nuwandavek can we confidently split that one out from the catch all and give more specific instructions?

jtexp, 2023-12-04:
> Looks like your steering alignment may be off.
> 
> @nuwandavek can we confidently split that one out from the catch all and give more specific instructions?

Ok thanks. It would be helpful to have a better error message for this. I think due to a steering wheel spacer I use my clockspring was about 15 degrees off from my wheel. I think this would make sense given the value of angleOffsetAverageDeg ?

Additionally I installed 0.9.4. This helped because it seems like in release/master the error message has regressed to paramsd. Before it would say "Vehicle parameter identification failed” which returns far more results.

Issue is sorted now. openpilot works as expected.

adeebshihadeh, 2023-12-04:
Yeah, I'll leave this open until we decide if we can make a specific error for this.

nuwandavek, 2023-12-04:
@jtexp yes, if the angleOffsetAverageDeg is very high (>10 deg), it usually points to a steering alignment issue.
@adeebshihadeh It usually manifests as the offset, but it can also show up in the other variables (depends on the init state, really). Need to do a TP/FP check before separating it out. I'd rather not cause more confusion, by claiming a wrong issue.

closed 2023-12-04 by nuwandavek