commaai/openpilot #30676: Quiet/distorted sounds

ErichMoraga, 2023-12-10:
### Describe the bug

After updating `master-ci`, I took my car for a drive, and noticed the engage/disengage sounds were quiet/distorted.

### Provide a route where the issue occurs

5e75f47a17ee2815|2023-12-10--12-55-21

### openpilot version

44b9c9b8eef69d1632a41653dd723cf6f3f7ad68

adeebshihadeh, 2023-12-18:
A bunch of soundd fixes have been pushed since. Have you noticed it since opening this?

ErichMoraga, 2023-12-18:
> A bunch of soundd fixes have been pushed since. Have you noticed it since opening this?

Tonight, I re-installed `master-ci`, then engaged/disengaged numerous times... no problem.  Looks like this GHI can be closed.

closed 2023-12-18 by adeebshihadeh