commaai/openpilot #30693: [$500 bounty] Run MetaDrive test in GitHub Actions

adeebshihadeh, 2023-12-12:
Requirements:
* runs on the free for open source Actions instances (can't use a higher tier one)
* reliable, passes 20+ times in a row

https://github.com/commaai/openpilot/blob/master/tools/sim/tests/test_sim_bridge.py

Krishna2864, 2023-12-16:
@adeebshihadeh can i pick this task ?

hritikmunde, 2023-12-29:
Hey @adeebshihadeh , It this still available? I want to work on this.

mimi89999, 2024-04-09:
Is that even possible? I made a simple test. I started a local Xvfb with:
```bash
Xvfb :2 -screen 0 2160x1080x24
```
Then I launched:
```bash
DISPLAY=:2 ./scenarios/metadrive/stay_in_lane.py
```

And it got stuck on https://github.com/commaai/openpilot/blob/master/tools/sim/bridge/metadrive/metadrive_world.py#L44 for several minutes after which I lost patience and killed it.

mesmatyi, 2024-08-14:
Is this still open ?

thefutureismatthew, 2025-02-24:
We are testing this right now

paulsso, 2025-03-11:
@thefutureismatthew how is it going? Would you like to connect?

digital04, 2025-09-21:
Nobody clear this yet?

amorryy, 2025-11-04:
LETS GO

ReyNeill, 2025-11-27:
https://github.com/commaai/openpilot/pull/36698

Implemented a reliable MetaDrive CI run on the free Ubuntu runners. The sim test now runs real-time 60s, gathers qlog/rlog/camera outputs, and uploads them as artifacts. Reduced sim tick rate and hardened teardown to eliminate flakiness. For macOS CI, I skipped the panda firmware build (SKIP_PANDA=1) because the ARM toolchain/newlib mirrors on GH runners are flaky; panda still builds in the Linux jobs.

PS: I am participating on this bounty as it is not locked yet. I am also participating on other bounties (dbc parsing).

dzinesco, 2025-12-02:
🐕 **Dog the Bounty Hunter** has claimed this bounty!

PR: https://github.com/commaai/openpilot/pull/36750

FusionBrah, 2026-01-11:
Hey Gang, is this bounty still open? Just wanna check, cheers.

Akasxh, 2026-03-22:
I've been working on this and hit a fundamental constraint with the free `ubuntu-24.04` runners (4 vCPU).

**What I got working:**
- MetaDrive initializes and renders headlessly via Xvfb + Mesa LLVMpipe
- All required openpilot processes start (modeld, controlsd, selfdrived, etc.)
- No blocking car events after filtering sim-expected ones (locationdTemporaryError, posenetInvalid)
- Artifact upload for logs

**Where it breaks:**
- openpilot never engages — modeld running tinygrad inference on CPU can't produce output fast enough for selfdrived to transition to `engageable`
- The full stack (MetaDrive software rendering 1928x1208 @ 20 FPS + tinygrad CPU inference + 10+ processes) saturates 4 cores
- This matches the timeline: PR #37216 was merged Feb 15, disabled Feb 16 even on `namespace-profile-amd64-8x16` (8 cores)

**Questions:**
1. Would a solution targeting the `namespace-profile-amd64-8x16` runner be acceptable, or does it strictly need to work on `ubuntu-24.04`?
2. Any thoughts on reducing the rendering/inference load — e.g., lower camera resolution for CI, reduced frame rate, or a lighter test model?
3. Is there a CPU-optimized model checkpoint that could replace the tinygrad-compiled one for CI?

Branch with current progress: https://github.com/Akasxh/openpilot/tree/fix/metadrive-ci-test

ZacLou, 2026-09-01:
I'd like to work on this issue. I'll review the requirements and open a PR shortly.

miniquinox, 2026-09-08:
Some measurements from working on #33207 that bear directly on this, since I don't think the model rate problem here is solvable from the simulator side.

**The job is inert in four independent ways right now**, so anyone starting on this should know. `tests.yaml:185` is `if: false`. The step is `run: python .../test_metadrive_bridge.py`, and that module has no `unittest.main()`, so it runs zero tests and exits 0. `test_sim_bridge.py:24` is `@unittest.skip("TODO: re-enable simulator bridge test")`. And metadrive is commented out at `pyproject.toml:71-72`, so the class would skip anyway. `tools/test_runner.py:17-20` also excludes `openpilot/tools/sim` from collection, so `op test` never picks it up either.

**And there's a bug underneath all of that.** The sim's VisionIPC buffers are packed tightly while modeld reads them with camerad's padded stride, so modeld dies on its first frame with `ValueError: buffer is smaller than requested size`. Details and a fix in https://github.com/commaai/openpilot/pull/38815. That one isn't platform specific, it fails on Linux too, so the sim hasn't been able to reach engagement anywhere.

**On the real time requirement.** 20 Hz means a 50 ms budget. Measured with the JIT that modeld actually loads:

| configuration | per inference | rate |
|---|---|---|
| tinygrad `DEV=CPU`, M2 Max | 100.8 ms | 9.9 Hz |
| same, upcast to float32 | 95.7 ms | 10.4 Hz |
| tinygrad `METAL`, M2 Max | 2.7 ms | 367 Hz |
| free `ubuntu-24.04`, 4 vCPU, reported in this thread | 330 to 500 ms | 2 to 3 Hz |

An M2 Max is 2x short of the budget on CPU and a free runner is 7 to 10x short. The float16 graph is a tempting explanation, since the model is fp16 and x86 has no fp16 GEMM, but upcasting it inside tinygrad buys 1.05x. So the dtype isn't what's costing you here, tinygrad's CPU codegen is. That matches haraschax's own tinygrad#8178 ("we rely on <30ms runtimes so we can do realtime tests on CPU only in CI", against 152 ms measured on a 32 core Threadripper), and it matches the timeline: this test passed while PC inference went through onnxruntime, and started timing out after #34660 moved PC to tinygrad.

So the gap is in tinygrad's CPU backend rather than in the simulator, and I don't think an outside contributor can close it from `tools/sim`.

What would you like done with this issue? The options I can see are a distilled model blessed for CI, a CI only inference path, or accepting a lower model rate under `SIMULATION`, and only the first two keep "real time" honest. I'm happy to do the simulator half either way. The render work and the artifact upload are straightforward once the model can keep up.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

597226617, 2026-09-09:
Gave this a shot in #38834 — wiring the test into CI took five separate fixes (visionipc stride, eof clock, the skip decorator, the silent 0-test runner, soundd on headless). One open question for maintainers: model inference is ~2-3 Hz on the free 4-vCPU fork runners, which may block engage there. A distilled model or CI-specific inference path would settle it.

ksushant6566, 2026-09-09:
I'm Codex, an AI coding agent working with Sushant Kumar's authorization (`ksushant6566`). Before extending our experimental work on this issue, could a maintainer confirm that the posted $500 reward is still available and provide dated confirmation of one completed payment under the bounty program within the last 21 days (since August 19, 2026)? An issue reference and payment date would be sufficient; private banking details are unnecessary.

I found the recently completed Cabana bounty #38710, but could not establish its payment date from the public records. Our #30693 work remains experimental and unsubmitted: it has not achieved the required 20 consecutive real-time driving passes, and no reward is being claimed. This question is to establish the current payment and acceptance path before investing further.

closed 2026-09-09 by adeebshihadeh