commaai/openpilot #30702: [$300 bounty] timed: get time and timezone from the modem

adeebshihadeh, 2023-12-12:
The comma 3/3X has a [Quectel EG25-G ](https://www.quectel.com/product/lte-eg25-g), and we should be able to pull the time and timezone from the cell network. If this ends up working well, this will remove the timezoned daemon, a bunch of panda RTC code, and dependence on an RTC battery.

The goal is to write a small Python daemon to get the current time and timezone. Requirements:
* works without a SIM, an inactive SIM, and an active SIM
* must replace the `timezoned` functionality
* must be fast (aside from modem bootup and ModemManager startup time)

dcondrey, 2023-12-13:
`
#!/usr/bin/env python3
import logging
import argparse
from serial import Serial, SerialException

def get_modem_timezone(serial_port, baudrate, timeout):
    """
    Retrieves the current time and timezone from the modem using AT+CTZR command.
    """
    try:
        with Serial(port=serial_port, baudrate=baudrate, timeout=timeout) as modem:
            # Sending command to enable extended time zone reporting and read current setting in one go
            modem.write(b'AT+CTZR=2;+CTZR?\r')
            response = modem.read_until(b'OK').decode()

            return response
    except SerialException as e:
        logging.error(f"Serial communication error: {e}")
        return None

def main():
    parser = argparse.ArgumentParser(description="Modem Time and Timezone Fetcher")
    parser.add_argument('-d', '--device', required=True, help='Specify device location, e.g., /dev/ttyUSB2')
    parser.add_argument('-b', '--baudrate', default=115200, type=int, help='Specify serial baudrate, default is 115200')
    parser.add_argument('-t', '--timeout', default=1, type=int, help='Specify serial timeout, reduced for faster response')
    args = parser.parse_args()

    time_and_timezone = get_modem_time_and_timezone(args.device, args.baudrate, args.timeout)
    if time_and_timezone:
        print(f"Time and Timezone: {time_and_timezone}")
    else:
        print("Failed to retrieve time and timezone from the modem.")

if __name__ == "__main__":
    logging.basicConfig(format='[%(levelname)s] %(message)s', level=logging.ERROR)
    main()
`

dcondrey, 2023-12-13:
I've developed a Python script that might help resolve the issue of retrieving time and timezone information from the Quectel EG25-G modem for the comma 3/3X. The script uses AT commands to communicate with the modem and fetches the required data. Here's the optimized version of the script:

```python
#!/usr/bin/env python3
import logging
import argparse
from serial import Serial, SerialException

def get_modem_time_and_timezone(serial_port, baudrate, timeout):
    """
    Retrieves the current time and timezone from the modem using AT+CTZR command.
    """
    try:
        with Serial(port=serial_port, baudrate=baudrate, timeout=timeout) as modem:
            # Sending command to enable extended time zone reporting and read current setting in one go
            modem.write(b'AT+CTZR=2;+CTZR?\r')
            response = modem.read_until(b'OK').decode()

            return response
    except SerialException as e:
        logging.error(f"Serial communication error: {e}")
        return None

def main():
    parser = argparse.ArgumentParser(description="Modem Time and Timezone Fetcher")
    parser.add_argument('-d', '--device', required=True, help='Specify device location, e.g., /dev/ttyUSB2')
    parser.add_argument('-b', '--baudrate', default=115200, type=int, help='Specify serial baudrate, default is 115200')
    parser.add_argument('-t', '--timeout', default=1, type=int, help='Specify serial timeout, reduced for faster response')
    args = parser.parse_args()

    time_and_timezone = get_modem_time_and_timezone(args.device, args.baudrate, args.timeout)
    if time_and_timezone:
        print(f"Time and Timezone: {time_and_timezone}")
    else:
        print("Failed to retrieve time and timezone from the modem.")

if __name__ == "__main__":
    logging.basicConfig(format='[%(levelname)s] %(message)s', level=logging.ERROR)
    main()
```

This script is optimized for faster execution and combines commands to reduce the number of read and write operations. It's designed to be run as a superuser and requires the `pySerial` library. Please test and adjust as necessary based on the modem's performance and response format.

adeebshihadeh, 2023-12-13:
This is a good start. Once it has the time and timezone, all it has to do is set the time like [this](https://github.com/commaai/openpilot/blob/fb977e45cdbc8852cb65f4aa153cb1e9a59accac/selfdrive/boardd/set_time.py#L29) and the timezone like [this](https://github.com/commaai/openpilot/blob/fb977e45cdbc8852cb65f4aa153cb1e9a59accac/system/timezoned.py#L19).

It also looks like this won't work with ModemManager running. AT commands can be run with `mmcli -m any --command=<cmd>`.

dcondrey, 2023-12-13:
Thanks.  I'll create a branch and see if I can get help testing.

References:
 - [Quectel_EC2x&EG9x&EG2x-G&EM05_Series_AT_Commands_Manual_V2.0.pdf](https://github.com/commaai/openpilot/files/13657694/Quectel_EC2x.EG9x.EG2x-G.EM05_Series_AT_Commands_Manual_V2.0.pdf)
 - [Quectel_Time_Synchronization_Introduction.pdf](https://github.com/commaai/openpilot/files/13657693/Quectel_Time_Synchronization_Introduction.pdf)

closed 2024-01-22 by adeebshihadeh