commaai/openpilot #30715: [$100 bounty] Setup a PR review bot

adeebshihadeh, 2023-12-13:
To start, I'm thinking the bot's review just consists of checking that a template is filled out in the PR body. 
* keep PRs as drafts until the bot is done with its "review"
* also add a special `bot-review` label while a PR is in the bot review phase
* must run in GitHub Actions

Set this up on your own fork of openpilot, then make a PR here.

amitsgh, 2023-12-14:
@adeebshihadeh, I would like to work on this. Could you provide additional details (if any) related to this?

AbhinavGoel9, 2023-12-14:
@adeebshihadeh, I'm interested in working on this. Could you provide any additional details?

mhardik003, 2023-12-23:
Can you assign this to me please?

Also could you please provide the template that you want to check along with it?

bilalshaikh292, 2023-12-23:
heyy @adeebshihadeh  is the issue still open ? I would like to contribute ?

hritikmunde, 2023-12-29:
Hey @adeebshihadeh , do you want any specific bot on this or anything that works?

umairjavaid, 2024-01-16:
Hi @adeebshihadeh, I just looked into the two PRs that is open and thought to share the next,he first PR is using a Python script to solve the issue and the second one is using the Action compose feature which allows you to run a bash script to do the work you need.

Have you thought to use open source action tools that could simplify the issue and make it easier? I was thinking to use one of the next:

https://github.com/marketplace/actions/template-validator
https://github.com/marketplace/actions/pr-description-checker

As for the label for bot review, I think just having a label at the end would be more logical since the bot review would take just a few seconds. WHat do you think?

I think my previous colleague @a0m0rajab could help and provide more insights as well.

You can add this into the comment, in issues, and have a part of it as PR review for both of the Opened PRs.

closed 2024-01-16 by jnewb1