commaai/openpilot #30784: "LTE" "CONNECT OFF:INE"

haram-KONA, 2023-12-18:
### Describe the bug

I am using "Prime Lite" and live in Korea. 
Since some time ago, the "LTE" connection status on the home screen appears to be normal, but the "CONNECT" status is maintained as "OFFLINE", and logs are not uploaded while driving. 
I don't know why. Before that, it was always connected and worked well.

### Provide a route where the issue occurs

1618132d68afc876

### openpilot version

0.9.6 master-ci

haram-KONA, 2023-12-18:
[image]

haram-KONA, 2023-12-18:
[image]

adeebshihadeh, 2023-12-18:
We do not provide support for getting your own SIM working with prime lite.

closed 2023-12-18 by adeebshihadeh