commaai/openpilot #30815: Intermittent "controls are lagging" and "Can Bus Disconnected" Error message - IONIQ 5

JJeonbear, 2023-12-20:
### Describe the bug

I drive a 2023 Ioniq 5, and since my fingerprint is not registered, I am using a fork to operate it.

Sometimes when I driving my car I get intermittent "controls are lagging" and "Can Bus Disconnected" error messages sometimes leading to a disengagement.
I have reloaded openpilot and still happens. I dont know how to reproduce the problem, but it has happened on everyday separate occasions.
There is no damage to the device always hangs on the window. Never took it apart either.

### Provide a route where the issue occurs

aad0fc7230a3066e|2023-12-15--15-56-50

### openpilot version

0.9.5

adeebshihadeh, 2023-12-20:
Can you connect your device to Wi-Fi to upload the full logs?

JJeonbear, 2023-12-21:
> Can you connect your device to Wi-Fi to upload the full logs?

aad0fc7230a3066e|2023-12-20--23-36-47

I drove again today to record the log.

I have uploaded the complete driving full log

JJeonbear, 2024-01-09:
> Can you connect your device to Wi-Fi to upload the full logs?

Did you check the log after uploading? The use of Comma 3x is not possible due to CAN bus disconnection.

adeebshihadeh, 2024-01-09:
I just added it to the milestone for the next openpilot release. I will look into it and hopefully fix for the next releaes.

JJeonbear, 2024-01-09:
> I just added it to the milestone for the next openpilot release. I will look into it and hopefully fix for the next release.

Thank you

royjr, 2024-02-07:
May be similar to https://discord.com/channels/469524606043160576/1152498009427419197/1204899591149981697

adeebshihadeh, 2024-05-17:
Tracking in #32286

closed 2024-05-17 by adeebshihadeh