commaai/openpilot #30834: Branch Switcher Loads Slow and freezes at "Finalize Update"

sheaduncan, 2023-12-21:
### Describe the bug

When choosing "Software" the data takes a bit of time to populate. Once it's populated and a new branch is selected and the download button is selected. It freezes on "Finalizing Update". This is on a home wifi connection with a fresh copy of OP master with only changes to hardcode fingerprint due to unsupported car.
[image]

### Provide a route where the issue occurs

09136c309ba9461d|2023-12-21--14-33-53

### openpilot version

0.9.6

adeebshihadeh, 2023-12-21:
“Finalizing” doesn’t have anything to do with your internet connection. It’s making a copy of the repo, so it can swap it atomically.

closed 2023-12-21 by adeebshihadeh

sheaduncan, 2023-12-21:
Still in "Finalizing Update"

sheaduncan, 2023-12-21:
Logs from trying to switch branches through SSH since the UI branch installer wouldn't work:

git fetch origin Ford-hysteresis-testing
remote: Enumerating objects: 90582, done.
remote: Counting objects: 100% (90582/90582), done.
remote: Compressing objects: 100% (29715/29715), done.
remote: Total 89380 (delta 63022), reused 84902 (delta 58549), pack-reused 0
Receiving objects: 100% (89380/89380), 397.11 MiB | 5.08 MiB/s, done.
Resolving deltas: 100% (63022/63022), completed with 934 local objects.
From https://github.com/sheaduncan/openpilot
 * branch                Ford-hysteresis-testing -> FETCH_HEAD
Fetching submodule cereal
From https://github.com/commaai/cereal
   8ae5c51..9bce87f  master             -> origin/master
 * [new branch]      pre-commit-updates -> origin/pre-commit-updates
   916f4a9..02c0254  stubs              -> origin/stubs
Fetching submodule opendbc
From https://github.com/commaai/opendbc
 * [new branch]      fix-subaru-throttle -> origin/fix-subaru-throttle
 * [new branch]      pre-commit-updates  -> origin/pre-commit-updates
Fetching submodule panda
From https://github.com/commaai/panda
 * [new branch]        2024-kia-sorento-hev   -> origin/2024-kia-sorento-hev
 + 2825df69...431f8b84 chrysler-long          -> origin/chrysler-long  (forced update)
 * [new branch]        gv70-2024-fw           -> origin/gv70-2024-fw
   5b41a63a..6e645d2d  master                 -> origin/master
 * [new branch]        nissan-cleanup-checks  -> origin/nissan-cleanup-checks
 * [new branch]        nissan-vel-mismatch    -> origin/nissan-vel-mismatch
   89abe134..69a28982  outback-forester-22-PR -> origin/outback-forester-22-PR
 * [new branch]        pre-commit-updates     -> origin/pre-commit-updates
 * [new branch]        subaru-fix-speed-units -> origin/subaru-fix-speed-units
   b37b6f23..3ad860b9  toyota-disable-radar   -> origin/toyota-disable-radar
Fetching submodule rednose_repo
Fetching submodule teleoprtc_repo
From https://github.com/commaai/teleoprtc
   4df237c..ca10fb2  master     -> origin/master
Fetching submodule tinygrad_repo
From https://github.com/geohot/tinygrad
 * [new branch]        fast_compare     -> origin/fast_compare
 * [new branch]        lazy_asserts     -> origin/lazy_asserts
   66f07d97..5cac6338  master           -> origin/master
 * [new branch]        revert-2625-upcast-phi-the-other-way-around -> origin/revert-2625-upcast-phi-the-other-way-around * [new branch]        revert-2792-uop-unary-check -> origin/revert-2792-uop-unary-check
 * [new branch]        revert-2801-test-randint -> origin/revert-2801-test-randint
 + df4f60aa...5dfcb7d2 update_benchmark -> origin/update_benchmark  (forced update)
From https://github.com/geohot/tinygrad
 * branch              2993dfe9219089bd1464741757f0c2ad67fe6a2e -> FETCH_HEAD
 * branch              46f6db75228cb7744a3a0a87616bfffeffa93658 -> FETCH_HEAD
 * branch              82ca9c66664b5f3739e1881f62cb94b72cf0b1fa -> FETCH_HEAD
 * branch              870ea766eec7a38d7d590c81436f15271ba2667e -> FETCH_HEAD
From https://github.com/commaai/rednose
 * branch            3a6f937ad1bb234dca7b0b2052faabdd58f3e085 -> FETCH_HEAD
 * branch            fbf65ab4aeec9a6891eaf22ddf430e592ab67aa4 -> FETCH_HEAD
From https://github.com/commaai/panda
 * branch              01df29e00b6d3dcceb7b4da18ddf97118382c659 -> FETCH_HEAD
 * branch              0b8c477302af4a10da92f5ba1d7af2a84e2794a7 -> FETCH_HEAD
 * branch              168b0db6a7ed53124a34ba2c2db03e7dd5b5dc26 -> FETCH_HEAD
 * branch              435cabe7f7a02a0c732725b97cbb7192a68a406d -> FETCH_HEAD
 * branch              47ee46540f4e2691fd2b390ecb3d83959eed215d -> FETCH_HEAD
 * branch              5440de0fdee3f04a1b1a15291854d2a183af5f67 -> FETCH_HEAD
 * branch              5984fd35096a9e3f9748435868ad49a199d4404d -> FETCH_HEAD
 * branch              7abca94c12ffdf2c65246d016af7f730f93d6bf1 -> FETCH_HEAD
 * branch              98ee810a0382e22290df2494b3877ed47d196230 -> FETCH_HEAD
 * branch              a32959accc12de3462e7446fd41e7dd79f6a9ec0 -> FETCH_HEAD
 * branch              a8917adcdf5b122ffd04f3734de9e0339531f181 -> FETCH_HEAD
 * branch              b29ae782f78be87e7ac1414773c6015b14eea728 -> FETCH_HEAD
 * branch              fbbc6b6996173573534613c7fce66db05a6aa721 -> FETCH_HEAD
From https://github.com/commaai/opendbc
 * branch            18bff2369b44ecb36638885c974aae1891616ae5 -> FETCH_HEAD
 * branch            48e3bba5ae2ecc63b1e00e1feaba55a7fa69b0b4 -> FETCH_HEAD
 * branch            a4407d6049c815f74e7225433353f2d22129cd03 -> FETCH_HEAD
[... 33 lines cut ...]
 * branch            e232a014579ba3a45bdba1968502c11ae2005f1a -> FETCH_HEAD
 * branch            e9d3597d23311a3e33a2def74ceec839e5ff4bf5 -> FETCH_HEAD
 * branch            efcbdb81ace2dfd52b9abf5099d416560a67c4e9 -> FETCH_HEAD
 
 git reset --hard FETCH_HEAD
HEAD is now at 99a8fadb9 Update interface.py

 git submodule update -f
Submodule path 'body': checked out 'cad7ef43ad5daf40b44d4792d47671c397cca376'
Submodule path 'cereal': checked out '8ae5c51a6515c171f3a82c9511a18d79a9e55c07'
Submodule path 'opendbc': checked out '93b983d49a2d6d5c67e15ce7650f55e4f121485d'
Submodule path 'panda': checked out '6e645d2d337f0c6b245ebae9bd8a54467634f13b'
Submodule path 'rednose_repo': checked out '44e8a891a2810f274a1fa980775155d9463e87b9'
Submodule path 'teleoprtc_repo': checked out 'ca10fb2410ee14e207f30aadebeec707904dc559'
Submodule path 'tinygrad_repo': checked out 'ae5d1407ee844a97a52ad3756835d38e7e2b9e1b'

 sudo reboot
 
git branch -a
* ford-tune-actuate-brake
  remotes/origin/ford-tune-actuate-brake
 

 
 For some reason it still fails