commaai/openpilot #30858: GM: Cruise Fault at standstill

morrislee, 2023-12-27:
### Describe the bug

Some cases, where I like to use stock ACC or OpenPilot ACC to hold my brake at a stop light, occasionally would trigger Cruise Faulted red alert.

I am able to replicate this on master, master-ci and other forks as well. Happens both on stock and op ACC.

[image]

### Which car does this affect?

2020 GMC Sierra 1500

### Provide a route where the issue occurs

dc7716b32bf25574|2023-12-22--21-09-03--5

### openpilot version

0.9.6

sshane, 2024-01-02:
- Is this a new issue, if you can recall?
- This happens on stock ACC? iirc, we found stock didn't let you enable at a stop, but openpilot could. Is that wrong?
- What are the conditions to cause this cruise fault? This happens every time? Or first time you stop? Or with pressing certain button combinations? Etc

morrislee, 2024-01-02:
This happens when I use both stock or op long, only happens when I am at a stop to attempt to use either stock or op acc to hold the brakes. Sometimes I can get this to happens a few times in a row at the same stop, but most of the time, I can wait for the message to clear is a good timing to try again and everything is fine as if nothing happened. This actually happened again on a fork garrettpall has me trying. 

I frequently use stock acc to hold the brakes, it has always worked this way at standstill, if speeds between 1 to 5 kph, I cannot enable stock acc, but op acc allows this, doesnt change the fact that this error happens in both stock or op long. 

I cannot find a pattern if there is a particular button I am pressing in combination or other conditions are met, it feels like a timing thing as if this shows up, it can do a few times or if I try right after it would work no problem. Its rare, but about 1 in 10 drives could happen.

Verylukyguy, 2024-01-03:
I just came back from on a drive on the newly updated Nightly branch and had three of these RSOD Faults.
df51b2195de66a9b|2024-01-02--19-41-45--0

morrislee, 2024-01-08:
another route, even with garrettpall helped me fix the Gas Regen Commands still happens

dc7716b32bf25574|2024-01-07--21-05-27--4
dc7716b32bf25574|2024-01-07--21-05-27--5

morrislee, 2024-01-21:
Happened on the la_model branch, this one was very different, I was able to trigger it while rolling, this might tell us more

dc7716b32bf25574|2024-01-20--17-51-35--7

garrettpall, 2024-01-22:
Besides the one that happened while moving, these appear to happen when no one is front of you and you remove your foot off the brake before the cruise state goes into standstill mode. This appears to be a stock issue and only becomes a problem because openpilot notifies you of it. I don’t think there is much that can be done to fix this. This issue also appears to have the symptom of releasing the brake hold when it faults.

morrislee, 2024-01-23:
anything we can do to catch this error and not throw a red alert?

sshane, 2024-05-13:
Stock ACC, ECM faulted on resume when the first engagement wasn't long enough (two frames): https://connect.comma.ai/dc7716b32bf25574/1715540216713/1715540294232

morrislee, 2024-05-13:
dc7716b32bf25574/00000068--88db216abb/1

morrislee, 2024-05-13:
dc7716b32bf25574/00000066--be6c18cfdc/5
dc7716b32bf25574/00000064--656e9cdcd3/25
dc7716b32bf25574/0000005d--737fd876c5/1

morrislee, 2024-05-17:
dc7716b32bf25574/2024-05-16--21-09-20/26

morrislee, 2024-05-17:
dc7716b32bf25574/2024-05-14--18-02-25/3

morrislee, 2024-05-24:
dc7716b32bf25574/2024-05-22--09-08-44/2