commaai/openpilot #30872: Mazda 2024 CX5 Dashcam Mode , Car Unrecognized

hongsrich, 2023-12-30:
### Describe the bug

After installing the release branch with all the wiring connected on my 2024 Mazda CX-5 Select, I am getting the issue Dashcam Mode Car Unrecognized.  I have tried uninstalling and trying again with the Master-CI branch but am still running into the same issue.  

### Provide a route where the issue occurs

https://connect.comma.ai/ad9d38124d3a3ef5/1703955559407/1703956202678

### openpilot version

0.9.5

### Additional info

Dongle ID:  ad9d38124d3a3ef5
Serial: f0014f8d

hongsrich, 2023-12-31:
Have also tried fingerprinting with the following code here, but running into the same issue

https://github.com/hongsrich/openpilot/edit/master/selfdrive/car/mazda/values.py#L11C0-L12C0

hongsrich, 2024-01-03:
Works now using the following branch:

https://github.com/TheOfficialJosh/openpilot/tree/24-mazda-cx5

closed 2024-03-12 by sshane

wenqian-ye, 2024-03-19:
I am facing the same problem. Is it merged in the latest openpilot 0.9.6?

sshane, 2024-03-20:
@wenqian-ye what is your dongle id?

wenqian-ye, 2024-03-20:
> @wenqian-ye what is your dongle id?

It is 833d1eb9ecc9b842

wenqian-ye, 2024-03-22:
@sshane Hi Shane, are there any updates? Is the system supported on the Mazda CX-5 2024 model?