commaai/openpilot #30880: KIA EV9 steering not working

benign07, 2024-01-01:
### Describe the bug

Happy new year everybody.
I bought KIA EV9 , and I am trying to use openpilot with c3x.
I success recognizing EV9 with sunnypilot without an error code, but there is a problem with steering(LKA not working), so I would like to ask for help. .

​Symptom
--> Even if you press the lane maintenance button on the steering wheel, a ringing sound and a blue border (mad mode) appear at the comma, but there is no response from the vehicle. (The green handle-shaped icon should turn on on the instrument panel, but it does not turn on and the function does not work.)

I have changed files while adding vehicles
-override.yaml: Add EV9 value
-values.py: Add carparam obtained from original dash cam
-interface.py: Add spec value
-sunnypilot_carname_json: EV9 car name
-​In addition to the above, other changes were made, such as addition to the CANFD list, EV list, etc.

P.S. Looking at the specifications again, I think it may be because LFA2 was introduced starting from EV9 and has not yet been reflected in Open Pilot(just guess). 
It is said that it is currently only applied to EV9, new Santa Fe, and GV80 FL.

I would like to know the cause and solution. Thank you!
​
Comma connet route which issue occured
1903aefb9b228a7c%7C2023-12-31--22-36-09

Comma connet route which obtain car parameters with original dashcam mode
1903aefb9b228a7c|2023-12-30--20-47-48
1903aefb9b228a7c|2023-12-31--09-36-32

### Which car does this affect?

KIA EV9 2023

### Provide a route where the issue occurs

1903aefb9b228a7c|2023-12-31--22-36-09

### openpilot version

0.9.6

sshane, 2024-01-03:
> I have changed files while adding vehicles

It sounds like you almost have the port finished, we just need to decipher a new steering message. Can you open a PR with all your changes to get us started?

benign07, 2024-01-07:
Sorry for late.

Does PR mean pull request?

I have submitted a pull request for changes to files(override, fingerprints, interface, and values) in openpilot's master branch.

I'm not a programmer but just a beginner, so I don't know if I did it well. If there is anything else I need to do, please let me know.

adeebshihadeh, 2024-01-27:
Closing, let's move the discussion to your PR.

closed 2024-01-27 by adeebshihadeh