commaai/openpilot #30883: [$100 bounty] Script for doing UI translations with an LLM

adeebshihadeh, 2024-01-02:
Use an LLM (either OpenAI API or something local) to generate UI translations: https://github.com/commaai/openpilot/tree/master/selfdrive/ui/translations.

* local is preferred if possible (GPU required is fine), but we can use GPT4 if the translations are actually better
* there should be a flag for either doing all the translations or just filling in the unfinished ones

hritikmunde, 2024-01-02:
Hi @adeebshihadeh , can you explain this a little bit more? What exactly do we need to implement?

meirdev, 2024-01-02:
@adeebshihadeh I uploaded an example of automatic translation (currently using OpenAI) from an xml language file: https://gist.github.com/meirdev/fb63baa36dcaceebe7c74db07e9c339e

sshane, 2024-01-02:
@meirdev would you like to open a PR?

closed 2024-01-06 by adeebshihadeh