commaai/openpilot #30894: [$500 bounty] Reduce kernel boot time to <1s on comma 3/3X

adeebshihadeh, 2024-01-03:
`systemd-analyze` reports this for the latest AGNOS:
```
Startup finished in 3.650s (kernel) + 7.343s (userspace) = 10.994s
graphical.target reached after 7.059s in userspace
```

Get the kernel part of that down to <1s in a self-contained way that doesn't delay the rest of the openpilot.

AGNOS is built with https://github.com/commaai/agnos-builder

andiradulescu, 2024-02-04:
I'm planning to do the following (most of them extracted from the [`boot-time-slides.pdf`](https://bootlin.com/doc/training/boot-time/boot-time-slides.pdf) from the documentation links):

- [ ] Measure and optimize the kernel initialization functions:
  - [ ] Analyze [`boot.svg`](https://github.com/commaai/agnos-builder/assets/76900/4b4caecf-095b-4016-ada2-f5eb15e3de46) from https://github.com/commaai/agnos-builder/issues/110#issuecomment-1925239170
  - [ ] Remove unnecessary functionality:
• Find which kernel configuration parameter compiles the code, by looking at the Makefile in the corresponding source directory.
  - [ ] Postpone:
• Find which module (if any) the function belongs to. Load this module later if possible.
  - [ ] Optimize necessary functionality:
• Look for parameters which could be used to reduce probe time, looking for the module_param macro.
• Look for delay loops and calls to functions containing delay in their name, which could take more time than needed. You could reduce such delays, and see whether the code still works or not.

- [ ] Cut out unused drivers.

- [ ] Reduce kernel size:
  - [ ] Compile everything that is not needed at boot time as a module.
  - [ ] Find the biggest symbols in the compiled kernel, for those which could be unnecessary, study the corresponding Makefile to see how not to compile them, if possible.

- [ ] Check which kernel compression option is the fastest (gzip/lzma/xz/lzo/lz4/zstd or uncompressed - _I love [zstd](https://lwn.net/Articles/827605/)_ but most likely lzo or lz4 is the winner, since pretty unclear if zstd support can easily be added and if the benefit is worth it).
- [ ] Deferring drivers and initcalls - If you can’t compile a feature as a module (e.g. networking or block subsystem), you can try to defer its execution.
- [ ] Turning off console output (Disable it by passing the `quiet` argument on the kernel command line).
- [ ] Scale the CPU frequency to maximum (or some thermally limited maximum) during boot. The faster the CPU, the faster the boot.

Documentation links:
- https://medium.com/smartmechatronics/embedded-linux-boot-time-reduction-eda5f5b4c014
- https://bootlin.com/doc/training/boot-time/boot-time-slides.pdf
- https://elinux.org/images/6/64/Chris-simmonds-boot-time-elce-2017_0.pdf
- https://www.e-consystems.com/articles/Product-Design/Linux-Boot-Time-Optimization-Techniques.asp
- https://www.toradex.com/blog/embedded-linux-boot-time-optimization
- https://source.android.com/docs/core/architecture/kernel/boot-time-opt

adeebshihadeh, 2024-02-04:
We already use `quiet` to get the initial speedup of like 11s -> 3s. I suspect we'll get the most savings from disabling and deferring. Anything openpilot doesn't use is fine to disable. Also keep in mind that deferring something critical but it costs time later in bootup doesn't count, but something non-essential to driving like WiFi is fine.

NSK-syam, 2026-09-03:
Implemented the comma 3X boot critical-path changes in:

- kernel: https://github.com/commaai/agnos-kernel-sdm845/pull/139
- AGNOS builder/userspace: https://github.com/commaai/agnos-builder/pull/621

The official `./build_kernel.sh` completed successfully from a clean case-sensitive checkout, including `comma_tizi.dtb`, `Image-dtb`, modules, signed boot image, and UAPI headers.

The dominant change brings tizi onto the existing minimal-audio path (secondary playback + tertiary two-channel capture), which previously reduced the corresponding mici sound init from about 1.7s to 0.2s. It also moves both tizi touchscreen probes asynchronous and removes the unused NFC NQ driver.

I do not have a comma 3X connected locally, so the final `systemd-analyze` measurement and speaker/mic/touch hardware checks still need to be run on-device.

ronakkasaram, 2026-09-29:
ready to work on this immediately