commaai/openpilot #30914: OP won't activate, “paramsd temporary error”

exik23, 2024-01-05:
### Describe the bug

Tried to use toyota-disable-radar on 2024 Rav4 hybrid, per Shane's request. The installation is fine and calibration is done. But the can't set cruise control, “paramsd temporary error”. 

Shane's note,. "the param estimator says roll is invalid, can you open a new issue with that route?"

### Which car does this affect?

Toyota Rav4 2024

### Provide a route where the issue occurs

061932dac503c0f7|2024-01-04--19-39-11--4

### openpilot version

toyota-disable-radar

sshane, 2024-04-17:
@exik23 does this still happen? any updates @nuwandavek?

sshane, 2024-08-20:
@exik23 does this still occur?

exik23, 2024-08-20:
Nope it’s fixed!Sent from my mobile device

closed 2024-08-20 by adeebshihadeh

Sanad2014, 2024-11-11:
I have the same problem with rav4 2024 hybrid