commaai/openpilot #30924: Hyundai Tucson Hybrid 2022 CAN error when going onroad

tylermilner, 2024-01-06:
### Describe the bug

Bought and installed a C3X for my girlfriend's mom in her Hyundai Tucson Hybrid 2022, but we are getting "CAN Error: Check Connections" error when starting the car and openpilot going from offroad to onroad. See route `4c55c9ec1741c686|2024-01-06--13-31-21--0` for an example. Currently on `devel` branch for stock openpilot `0.9.5`.

I was able to walk her through the process of setting up her Connect account and sharing her device with me. I was able to SSH into the device via Connect and `tmux a` while she started the car. It looks like there might be some issue with the `ACCELERATOR_BRAKE_ALT` CAN message as the following errors started popping up constantly:
```
opendbc/can/parser.cc: 0x100 'ACCELERATOR_BRAKE_ALT' NOT SEEN
```

Screenshots of logs below (note there is some overlap between the screenshots to make sure I captured it all):
[image]
[image]

### Which car does this affect?

Hyundai Tucson Hybrid 2022

### Provide a route where the issue occurs

4c55c9ec1741c686|2024-01-06--13-31-21

### openpilot version

0.9.5

tylermilner, 2024-01-06:
I switched her to `master-ci` and tried again, but getting the same issue.

This appears to be a potential fingerprinting issue where the car is being detected as the ICE variant rather than the hybrid (e.g. seems to fingerprint as `TUCSON_4TH_GEN` rather than `TUCSON_HYBRID_4TH_GEN`). By reading the logs in Connect Admin, I was able to get these firmwares:
```
'carFw': [{'address': 1975,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'cornerRadar',
                          'fwVersion': b'\xf1\x003D',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1983,
                          'subAddress': 0},
                         {'address': 1971,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'hvac',
                          'fwVersion': b"\xf1\x00NX4   97255-P0030CONTROL ASS'Y-DATC  1.04 NX4HEV ATC 0.7 V1.00",
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1979,
                          'subAddress': 0},
                         {'address': 2000,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00NX4__               1.00 1.00 99110-N9100         ',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 2008,
                          'subAddress': 0},
                         {'address': 1988,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00NX4 FR_CMR AT USA LHD 1.00 1.00 99211-N9210 14G',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x00'],
                          'responseAddress': 1996,
                          'subAddress': 0},
                         {'address': 1969,
                          'brand': 'hyundai',
                          'bus': 0,
                          'ecu': 'parkingAdas',
                          'fwVersion': b'\xf1\x10NX4H ADAS_PRK ANL 1.00 1.01 99910-P0800',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'"\xf1\x10'],
                          'responseAddress': 1977,
                          'subAddress': 0}],
```

Based on the current `selfdrive/car/hyundai/values.py`, the `Ecu.fwdRadar` value of `b'\xf1\x00NX4__               1.00 1.00 99110-N9100         '` already exists, but I **did** need to add the value of `b'\xf1\x00NX4 FR_CMR AT USA LHD 1.00 1.00 99211-N9210 14G'` to `Ecu.fwdCamera`.

However, now the same `Ecu.fwdCamera` and `Ecu.fwdRadar` values appear in both `CAR.TUCSON_4TH_GEN` and `CAR.TUCSON_HYBRID_4TH_GEN`. I'm guessing it's defaulting to `CAR.TUCSON_4TH_GEN` since it appears first in the list and already has the matching firmwares:
[image]

For now, I will look into how to force the `TUCSON_HYBRID_4TH_GEN` fingerprint, but what's the best way to fix this going forward? Do we need to add additional ECU firmware types such as `cornerRadar`, `hvac`, or `parkingAdas` so that there is more information available to differentiate between the ICE and hybrid versions?

tylermilner, 2024-01-07:
After hard-coding to `TUCSON_HYBRID_4TH_GEN`, we were able to get everything working! ([is this the best way to hard-code?](https://github.com/tylermilner/openpilot/commit/647b8a6c946ad1cc35c948e8dc0fcd442511a2ab)) Still looking for guidance on how to upstream her `fwdCamera` firmware version into `master` so that there's not a collision with `TUCSON_4TH_GEN` though.

sshane, 2024-01-31:
Hey! We just merged a PR to detect if the car is hybrid or not, so your car will be supported on master after https://github.com/commaai/openpilot/pull/31238

closed 2024-01-31 by sshane