commaai/openpilot #30932: 2020 Subaru Forester: Check engine (code P26A6) and Eyesight problem; remove all hardware, all is fine.

mikelee1701, 2024-01-07:
### Describe the bug

2020 Subaru Forester: Check engine (code P26A6) and Eyesight problem; remove all hardware, all is fine.

Just got my Comma 3 and installed it without initial problems.  Drove it about 30 miles and all of sudden check engine light Comes on, Eyesight reports problems, and "S" blinking.  Removed all devices and restored car to factory hookups, and all returns to normal.  Re-installed, and I get same check engine, etc again.  Removed and all is fine again.

I am latest release, and i'm not even sure it's SW related.  As long as its plugged in it get issues.

### Which car does this affect?

2020 Subaru Forester

### Provide a route where the issue occurs

041e083d86170f22|2024-01-06--20-16-20--0

### openpilot version

0.9.5

mikelee1701, 2024-01-07:
I decided to go back out to start the car one more time and the check engine light came back with all comma hardware detached.  It seems I have another issue.  closing this for now.

closed 2024-01-07 by mikelee1701

jnewb1, 2024-01-08:
We actually had the exact same error code on our Subaru, and it turns out it was a known issue that the dealership fixed for free

mikelee1701, 2024-01-08:
Yes, just came back from dealership.  It was the Thermal control valve, it seems to be a "common issue" from searching the web.  I'll be re-installing the Comma 3.  What timing.  We just got the car too.