commaai/openpilot #31000: openpilot abruptly restarted shortly after starting car

goodchoom, 2024-01-14:
### Describe the bug

Upon starting my car for my morning commute, openpilot abruptly restarted shortly after going onroad. The car's dash then lit up with various check system warnings. After openpilot restarted it threw the **Cruise Fault: Restart car to engage** alert. After restarting my car I was able to engage and use openpilot with no apparent issues.

### Provide a route where the issue occurs

b76c43f138cb09ac|2024-01-14--07-06-34--0

### openpilot version

28215f659b56b49fa11fc4c673b191dafc701794

goodchoom, 2024-01-18:
It just happened again this morning. This time, it occurred several minutes into the drive.

adeebshihadeh, 2024-02-20:
Same as your other issue

closed 2024-02-20 by adeebshihadeh