commaai/openpilot #31007: [$500 bounty]

adeebshihadeh, 2024-01-15:


skushagra9, 2024-01-16:
hey , I would like to work on this

jakethesnake420, 2024-01-28:
Gonna give it a go https://github.com/commaai/openpilot/pull/31197

jakethesnake420, 2024-06-02:
I found this project https://github.com/parkouss/funq.
It works similar to QT Squish in that it doesn't rely on hard coded coordinates. The issue is the project is dead so we would need to add some things to locate some things. If we fork it and add the missing requirements then you could write python tests like this:

```python
from funq.client import FunqClient

funq = FunqClient("localhost", 9000)
wl = funq.widgets_list(with_properties=True)
settings_btn = funq.widget(alias="SidebarSettingsBtn")
settings_btn.click()
```

adeebshihadeh, 2024-06-23:
@thenhnn sorry for the delayed reviews. The blocker this time was dealing with where to push the screenshots since GitHub doesn't allow us to directly upload them. I just looked into this, and our two best options are a separate GitHub repo or I setup an Azure bucket.

I propose we go the GitHub repo route since it's at least not a separate piece of infrastructure. You can setup your own GitHub repo to develop with, then I'll switch it to one on the commaai org once it's ready to merge.

jakethesnake420, 2024-06-23:
In already have an example of posting to another repo. That was my first implementation. I stopped working on it because of this comment https://github.com/commaai/openpilot/pull/31198#issuecomment-1918152300 and i knew there was no better way.  here is an example of the link that would be posted with the PR https://jakethesnake420.github.io/openpilot/pr-c15bc3de40c8b21dffd761b230055cc628d3a274/index.html.

I can finish the implementation if you lock it to me again.

closed 2024-06-25 by adeebshihadeh