commaai/openpilot #31125: [100$ bounty] get metadrive test to pass in CI

jnewb1, 2024-01-23:
the metadrive test runs okay over here, but it fails about 1/20 times: https://github.com/commaai/openpilot/pull/30775

this bounty is to get this test to work 100% of the time in github actions.

requirements:
   - no services can be blocked, i.e. soundd and ui, (except the ones here that are normally blocked in the sim) https://github.com/commaai/openpilot/blob/ea57336005f4cf9c587dfbe221662b4de73bd508/tools/sim/launch_openpilot.sh#L9
   - must past 50 times in CI. you can use a matrix to test lots of runs at once, though I would wait until you have it running consistently locally https://github.com/commaai/openpilot/blob/3001d67e64f22bff0fe9cbc8f29a51aa00ab80e4/.github/workflows/tools_tests.yaml#L67-L70
 
this is separate from the main bounty, which would require it to run well https://github.com/commaai/openpilot/issues/30693

jakethesnake420, 2024-01-27:
Here it passed 50 times
 
 https://github.com/jakethesnake420/openpilot/actions/runs/7671252393

skushagra9, 2024-01-28:
Hey, I am interested to solve this

closed 2024-05-05 by adeebshihadeh