commaai/openpilot #31127: Device Offline in Comma Connect App

indyemmett, 2024-01-23:
### Describe the bug

Current member of Comma Prime Lite.  The connect app recognizes the Comma 3x and provides logs and history of drives.  When I enter a destination to use the navigation map feature, I get the following error: "device offline destination will be set once device is online."  Even when connected to a shared wifi, I am unsuccessful in connecting. 

### Provide a route where the issue occurs

https://connect.comma.ai/48629771f22c766b/1706014357663/1706015200976

### openpilot version

0.9.5

adeebshihadeh, 2024-01-23:
In the route you linked, it doesn't connect to Wi-Fi until the very end of the drive.

indyemmett, 2024-01-23:
I cannot use any of my navigation features and I cannot connect to my 3X using using the connect app to set a destination.

adeebshihadeh, 2024-01-23:
With prime lite, you provide your own internet connection, which the navigation features need.

indyemmett, 2024-01-24:
Thanks, but even on a shared wifi connection I still cannot connect. I had limited success for about 1 minute today after many failed attempts. I was able to set a destination but after I left the property it started to reroute and stopped working.

adeebshihadeh, 2024-01-24:
Navigation currently requires an active internet connection while driving to reroute and download map tiles. Prime lite is primarily offered for extended drive storage, SSH access, and for users who already have their own SIM card in the device. I'd recommend switching to full prime or cancelling your lite subscription if you are interested in the navigation feature.

closed 2024-01-24 by adeebshihadeh