commaai/openpilot #31150: Subaru: FW query inconsistent

sshane, 2024-01-25:
Brought to our attention from the FW bot, many Subaru's return different results for the same ECU and request:

```python
1c451e9ed3c48243|2024-01-15--16-10-38:
[('ecu', 'fwdCamera'), ('fwVersion', b'\x00\x00e\x9a\x00\x00\x00\x00\x00\x00'), ('address', 1927), ('subAddress', 0), ('responseAddress', 1935), ('request', [b'"\xf1\x82'])
vs
1c451e9ed3c48243|2024-01-15--20-36-22:
[('ecu', 'fwdCamera'), ('fwVersion', b'\x00\x00e\x9a\x00\x1f@ 1\x00'), ('address', 1927), ('subAddress', 0), ('responseAddress', 1935), ('request', [b'"\xf1\x82']) (edited) 
```

Race condition on startup?

jyoung8607, 2024-01-25:
Are they known to be doing this *currently*?

There used to be a similar problem with HKG. IIRC we weren't correctly implementing some aspect of ISO-TP flow control and so we'd intermittently see garbage past the first frame. There could be leftover artifacts in any port's FW from that time period.

sshane, 2024-01-25:
I believe it's the latest version, we'll try to reproduce on our Subaru soon. I believe this occasionally still happens for Hyundai's transmission FW but we don't need to rely on it so we're fine.

jnewb1, 2024-02-22:
happened on our outback here: ```344c5c15b34f2d8a/2024-01-26--09-09-29```

so we might be able to replicate this and find the cause

sshane, 2024-03-07:
Here's a case where a user's fwdCamera was returning `b'\r!\x08\x017\x00\x00\x00\x00\x00',` for 20 days until it switched to `b'\r!\x08\x017\n!\x08\x003',` and then we fingerprinted

2ef74f5bd24ef68b

sshane, 2024-03-09:
After 10 seconds our Outback's fwdCamera returns the full non-zero-suffixed FW. `b'\x00\x00eJ\x00\x00\x00\x00\x00\x00'` -> `b'\x00\x00eJ\x00\x1f@ \x19\x00'`