commaai/openpilot #31164: Getting an "Controls Unresponsive" "Reboot Device" error almost every time I start my car for the first time in the morning

dukeajuke, 2024-01-25:
### Describe the bug

Car - 2023 VW Arteon. I've had my Comma3 for about 14 months now, and I started getting an "Controls Unresponsive" "Reboot Device" error almost every time I start my car for the first time in the morning. Rebooting it sometimes fixes it temporarily, sometimes I have to unplug the power completely and let it boot again. Started happening about a month ago. Doesn't matter if the device is cold or warm.

### Provide a route where the issue occurs

c34ef43981b6d888|2024-01-25--14-15-03

### openpilot version

0.9.5 (Latest)

jyoung8607, 2024-01-26:
I looked into this in #hw-three-3x before the issue was opened. controls didn't crash. At the moment of startup, it appears USB connectivity bounced for both the modem and onboard Panda. Both quickly recovered, and boardd reconnected to Panda, but I think it may have hung controls startup by disrupting the boardd/controlsd startup/fingerprinting safety-setter sequence.

I'm not sure what could cause the USB disruption.

1okko, 2024-02-05:
I also encountered the same problem, but when I disconnected the ublox GPS module, everything returned to normal. You can try it out.

closed 2024-06-30 by adeebshihadeh