commaai/openpilot #31219: openpilot froze and restarted after resuming from standstill

goodchoom, 2024-01-29:
### Describe the bug

Shortly after resuming from a standstill, openpilot abruptly froze for about a minute before restarting. While the UI was frozen, openpilot emitted both the soft and immediate warning sounds.

After it restarted on its own, openpilot was unavailable for about a minute due to low communication rate between processes, but appeared almost normal, apart from the missing green navigation path.

openpilot appeared to work normally for several minutes until it threw a low memory OSOD, forcing me to disengage, then threw several alerts all at once, including driving model lagging, camera malfunction, and cruise fault. Afterwards openpilot was once again unavailable for the rest of the drive.

### Provide a route where the issue occurs

b76c43f138cb09ac|2024-01-29--06-59-20--0

### openpilot version

84df644123affc4234ca2054bdacb7b85cd33797

jyoung8607, 2024-01-30:
Some sort of event occurred with the onboard storage starting at 346629 and ongoing:

```
[346629.090972] MAIN 0 kernel - ufshcd-qcom 1d84000.ufshc: ufshcd_abort: Device abort task at tag 15
[346629.091033] MAIN 0 kernel - sd 0:0:0:0: [sda] tag#15 CDB: Write(10) 2a 00 01 64 46 28 00 00 80 00
[346629.091098] MAIN 0 kernel - ufshcd-qcom 1d84000.ufshc: ufshcd_print_fsm_state: TX_FSM_STATE = 5, err = 0
[346629.091166] MAIN 0 kernel - ufshcd-qcom 1d84000.ufshc: ufshcd_print_fsm_state: RX_FSM_STATE = 3, err = 0
[346629.091224] MAIN 0 kernel - 1d84000.ufshc host regs: 00000000: 1587031f 00000000 00000210 00000000
[346629.091276] MAIN 0 kernel - 1d84000.ufshc host regs: 00000010: 01000000 00010217 00000c01 00000000

(plus quite a lot more continuing UFS and SCSI subsystem errors and a kernel WARNING)
```

This was clearly blocking disk I/O for some period of time, see for example here where we're probably unwinding from an I/O stack-up. These reports should be coming out once per second but there's a large burst at 346641 with overages.

```
[346641.276239] system/loggerd/loggerd.cc:161 - handle_encoder_msg: rotate 11 -> 21 ready 1/4 for qRoadEncodeData
[346641.276300] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'qRoadEncodeData' messages
[346641.276354] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'qcomGnss' messages
[346641.276404] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'radarState' messages
[346641.276452] system/loggerd/loggerd.cc:275 - loggerd_thread: 739000 messages, 1094.51 msg/sec, 641.50 KB/sec
[346641.276500] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'roadCameraState' messages
[346641.276548] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'sendcan' messages
[346641.276592] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'uiPlan' messages
[346641.276638] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'wideRoadCameraState' messages
[346641.276685] system/loggerd/loggerd.cc:275 - loggerd_thread: 740000 messages, 1095.98 msg/sec, 641.97 KB/sec
[346641.276732] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'accelerometer' messages
[346641.276778] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'androidLog' messages
[346641.276824] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'cameraOdometry' messages
[346641.276891] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'can' messages
[346641.276949] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'carControl' messages
[346641.276998] system/loggerd/loggerd.cc:275 - loggerd_thread: 741000 messages, 1097.45 msg/sec, 642.85 KB/sec
[346641.277044] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'carState' messages
[346641.277090] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'controlsState' messages
[346641.277135] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'driverCameraState' messages
[346641.277180] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'driverMonitoringState' messages
[346641.277226] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'driverStateV2' messages
[346641.278723] system/loggerd/loggerd.cc:275 - loggerd_thread: 742000 messages, 1098.92 msg/sec, 643.29 KB/sec
[346641.283469] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'gyroscope' messages
[346641.286587] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'liveLocationKalman' messages
[346641.287047] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'liveParameters' messages
[346641.290870] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'liveTracks' messages
[346641.291308] system/loggerd/loggerd.cc:275 - loggerd_thread: 743000 messages, 1100.38 msg/sec, 644.20 KB/sec
[346641.294091] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'logMessage' messages
[346641.294620] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'longitudinalPlan' messages
[346641.295123] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'magnetometer' messages
[346641.323783] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'modelV2' messages
[346641.324555] system/loggerd/loggerd.cc:275 - loggerd_thread: 744000 messages, 1101.81 msg/sec, 647.12 KB/sec
[346641.326428] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'qRoadEncodeData' messages
[346641.329952] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'radarState' messages
[346641.330870] system/loggerd/loggerd.cc:280 - loggerd_thread: large volume of 'roadCameraState' messages
```

goodchoom, 2024-02-06:
Happened again this morning.

`b76c43f138cb09ac|2024-02-06--07-06-04--11`

goodchoom, 2024-02-18:
Encountered this again, but this time openpilot did not alert me of low memory.

`b76c43f138cb09ac/2024-02-18--16-32-54`

goodchoom, 2024-02-19:
Low Communication Rate between Processes/Communication Issue Between Processes: deviceState managerState

`b76c43f138cb09ac/2024-02-19--16-32-35/0`

adeebshihadeh, 2024-02-19:
I suspect this is a software issue, but one that will be difficult to fully root cause and fix remotely (might be a UFS driver bug). Let's swap your device out (comma.ai/support), and I'll look into it more once it's back here. IIRC you might be out of warranty, but that's fine since I'd like to see this device.

closed 2024-02-19 by adeebshihadeh