commaai/openpilot #31223: [100$ bounty] test more LogReader scenarios

jnewb1, 2024-01-30:
```auto_source``` and ```auto_fallback``` have introduced quite a few cases into logreader that can easily break, we want to test each one with a unit test in [```test_logreader```](https://github.com/commaai/openpilot/blob/master/tools/lib/tests/test_logreader.py). All api queries and downloading should be mocked to make this a proper unit test with no network or specific route dependence.

grottiTT, 2024-01-30:
I don't see `def auto_fallback` in `logreader.py`; I checked the most recent master and I don't see it in there.

Can you please explain what you meant by that?

jnewb1, 2024-01-30:
> I don't see `def auto_fallback` in `logreader.py`; I checked the most recent master and I don't see it in there.
> 
> Can you please explain what you meant by that?

meant auto_strategy, fixed

grottiTT, 2024-02-05:
For the second scenario, where we're passing a qlog file, is the `segment_range` found in the `QLOG_FILE` constant? This constant is at the top of the unit test file.
I see "0375fdf7b1ce594d/2019-06-13--08-32-25/3" which means 3 segments from this specific route?

@jnewb1 idk if my question makes sense, but we need to pass a SegmentRange to the `auto_source` function.

jnewb1, 2024-02-05:
A SegmentRange can be created from a string easily with

> SegmentRange('0375fdf7b1ce594d/2019-06-13--08-32-25")

adeebshihadeh, 2024-06-10:
Still interested, but closed due to low quality PRs. If someone wants to make a high quality PR, bounty is still up for grabs.

closed 2024-06-10 by adeebshihadeh