commaai/openpilot #31244: please feel free to re-open with a route so we can check this out

richardalexdominguez, 2024-01-31:
please feel free to re-open with a route so we can check this out

_Originally posted by @sshane in https://github.com/commaai/openpilot/issues/31006#issuecomment-1918317971_

richardalexdominguez, 2024-01-31:
Wish I knew how to put a route in here for you. Any steps I can take to get Dashcam mode to recognize my car? Would love to see what comma 3X can help me do…

sshane, 2024-01-31:
In the device settings it will have a dongle ID. You can post that here

richardalexdominguez, 2024-01-31:
It has the dongle ID: 54f2cff3e445c51e 
On the 3X but I see no way to download drives… 
Can you send a screenshot or more detailed directions?

johnnathanwood, 2024-01-31:
> It has the dongle ID: 54f2cff3e445c51e On the 3X but I see no way to download drives… Can you send a screenshot or more detailed directions?

You can find the drives on connect.comma.ai

### 1. Click on a drive and you will see two menu items (Select menu item ```More info```:
[image]

### 2. Select the preserve toggle 

[image]

[image]

### 3. Select menu item ```Files```

### 4. Upload all logs (Make sure your comma device is connected to wifi)

[image]
[image]

sshane, 2024-02-01:
Is there a reason you tried to fingerprint as a Civic Bosch?

sshane, 2024-02-01:
Please try this branch and let me know if it works:

Installer URL: https://installer.comma.ai/commaai/civic-bosch-no-obd

richardalexdominguez, 2024-02-01:
I did not intend for to fingerprint as a civic... 

YES the civic-bosch-no-obd branch is working. Thank you.

Are there any next steps you recommend so that I can take full advantage of the tech?

sshane, 2024-02-03:
Make sure you put your device on WiFi so we can check out the logs.

richardalexdominguez, 2024-02-03:
Whenever the car is home it connects to WiFi.

sshane, 2024-02-10:
Done in https://github.com/commaai/openpilot/pull/31259!

I noticed a few routes your camera and radar failed to respond (each in separate routes). Can you let me know when the device is on WiFi again?

closed 2024-02-10 by sshane