commaai/openpilot #31246: Hyundai Driver Grasp Warning Fault

johnnathanwood, 2024-01-31:
### Describe the bug

I have been encountering repeated grasp warnings under various conditions and routes

### Which car does this affect?

2022 Hyundai Kona ICE 

### Provide a route where the issue occurs

0099bdb24d82951b|2024-01-27--13-43-20--23

### openpilot version

0.9.6

### Additional info

Hardware:

1. Comma 3X

2. harness Type: [Hyundai O](https://github.com/commaai/openpilot/wiki/Hyundai-Kia-Genesis#harness-guide)

Vehicle Specs:

1. architecture: CAN

2. SCC: Radar/CAM SCC

Pre-Route Preperations:

1. Recalibration

2. Comma pluggin in before vehicle start

Additional Route(s):

```0099bdb24d82951b|2024-01-10--07-44-32--59```

```0099bdb24d82951b|2024-01-08--07-47-39--55```

johnnathanwood, 2024-02-03:
@sshane can we add this to the status of new HKG discussion for tracking purposes?

johnnathanwood, 2024-02-07:
Grasp warning
Route Id: 0099bdb24d82951b|2023-12-22--12-01-16--14
Route Id: 0099bdb24d82951b|2023-12-24--18-28-54--7
Route Id: 0099bdb24d82951b|2023-12-24--18-28-54--11

### Update: It appears to only happen when I start my vehicle and then plug in the comma. If I plug the comma in before, I do not get the warning during my routes

### Update 20240105: Did two 30 minute routes without getting the grasp warning with the comma plugged in before starting the car

Update 20240108: Car cutting in triggered the grasp warning
Route Id: 0099bdb24d82951b|2024-01-08--07-47-39--55

Update 20240110
Pre-Route Preparations:

Recalibration

Comma plugged in before vehicle start

In-Route actions:

Adjusted follow distance
Route Id: 0099bdb24d82951b|2024-01-10--07-44-32--59

Update 20240130
Pre-Route Preperations:

Recalibration

Comma pluggin in before vehicle start

Route id: 0099bdb24d82951b|2024-01-30--08-31-19--20

Time:  08:51:30

sshane, 2024-02-08:
From the first route I see the driver awareness status message fall to 1, and then you take over, is that when it started showing the alert?

[image]

johnnathanwood, 2024-02-08:
@sshane yeah that's when I take control. Also, yes, it appears that when I plug the comma in before starting the car I don't get the message

johnnathanwood, 2024-02-09:
@sshane that is when the alert happened on the dash, with audible alert, and acceleration paused. No alert on the comma when that happens. Also, I can confirm that when I plug the comma in and wait for it to initialize before starting the car, it 100 percent does not happen. It appears to only happen when the comma is never unplug and on standby when starting the car.

royjr, 2024-02-11:
Possibly related, 2024 Hyundai Kona ICE also has grasp warnings on https://github.com/commaai/openpilot/pull/30679 on seemingly random parts of the drive. Back to back warninigs sometimes then nothing for a while. Usuable, just very annoying.