commaai/openpilot #31292: LKAS fault / CAN error on every 1st drive of day

siddchou, 2024-02-03:
### Describe the bug

I see LKAS fault on every 1st drive of the day or if I wake up car after 2 to 3 hours .  

I see a popup at my car dashboard which says smart braking disabled camera fault . I some time get red error message on comma screen with can error and some times LKAS fault please restart the car . 

To be very honest the issue is very irritating even if I restart car at traffic stop it does not  away car needs to be stopped for 4 to 5 mins I need to get out and lock the car.  Then only it  goes away

### Which car does this affect?

Mazda cx5 2024 

### Provide a route where the issue occurs

e548ddc194075bfb|2024-02-02--09-40-59--0

### openpilot version

0.9.5

### Additional info

Using latest branch

sshane, 2024-02-06:
@siddchou can you try a new USB C cable? Near the end of the route, while moving, the comma three thinks the car turned off because all of the CAN messages time out, and panda reports a lot of errors. This could be caused by a faulty cable.

siddchou, 2024-02-06:
I currently came on vacation and would be returning by 12th March . I can check once back.

Please check if below aswell is the same issue

e548ddc194075bfb|2024-01-31--16-12-06--0

siddchou, 2024-02-06:
Also same thing happened at 31st Jan aswell at 4 pm ast .  I was not able to resolve LKAS error even after switching  off the car 

https://useradmin.comma.ai/logs?url=https://commadata2.blob.core.windows.net/bootlogs/e548ddc194075bfb/2024-01-31--16-05-34.bz2?se%3D2024-02-06T18%253A37%253A06Z%26sp%3Dr%26sv%3D2018-03-28%26sr%3Db%26rscd%3Dattachment%253B%2520filename%253De548ddc194075bfb_2024-01-31--16-05-34.bz2%26sig%3DTyzy%252BDyCG2m8t7XH1BdbM6MrMzpfGWlkgD2LlvwPEBs%253D

https://useradmin.comma.ai/logs?url=https://commadata2.blob.core.windows.net/bootlogs/e548ddc194075bfb/2024-01-31--16-19-23.bz2?se%3D2024-02-06T18%253A37%253A06Z%26sp%3Dr%26sv%3D2018-03-28%26sr%3Db%26rscd%3Dattachment%253B%2520filename%253De548ddc194075bfb_2024-01-31--16-19-23.bz2%26sig%3DHl7sA03pwUOK/%252B4iO677nBgGxd%252BsGm/zLBoK8ygpdYU%253D

https://useradmin.comma.ai/logs?url=https://commadata2.blob.core.windows.net/bootlogs/e548ddc194075bfb/2024-01-31--16-19-23.bz2?se%3D2024-02-06T18%253A37%253A06Z%26sp%3Dr%26sv%3D2018-03-28%26sr%3Db%26rscd%3Dattachment%253B%2520filename%253De548ddc194075bfb_2024-01-31--16-19-23.bz2%26sig%3DHl7sA03pwUOK/%252B4iO677nBgGxd%252BsGm/zLBoK8ygpdYU%253D

sshane, 2024-03-12:
@siddchou have you tried a new OBD-C cable since then?

siddchou, 2024-03-12:
Hi @sshane   i tested it . I got same Can error on the very 1st drive . 2nd drive was  ok  . 

I will test it again

siddchou, 2024-03-13:
i tried today same thing happened on 1st drive of the day 

e548ddc194075bfb/2024-03-13--07-39-07/0

e548ddc194075bfb/2024-03-13--07-49-23/0

sshane, 2024-03-20:
Can you upload those routes and preserve them for when I have time to check this out?

siddchou, 2024-03-20:
@sshane  Hi I have uploaded all logs and preserved them. I don't have comma connect not sure till when the logs will be avaiable

sshane, 2024-03-20:
I would contact support here: https://comma.ai/support, this does seem unlikley to be a software issue.

You can see that on startup the CAN ignition signal dropped out, with other messages also being dropped with many panda errors.

[image]
[image]

closed 2024-03-20 by sshane