commaai/openpilot #31311: `commIssue` blocking entry on almost every car start with high CPU usage

sunnyhaibin, 2024-02-06:
### Describe the bug

On almost every car start after openpilot is initialized, openpilot cannot be engaged and blocked by `commIssue` NoEntry event. `commIssue` would usually stop after about 30 seconds or so. Interestingly, the first car start after the device is freshly booted does not experience this issue as severely.

CPU usage is also relatively high on core 2 and core 4.

**Routes**
- `2bbe12792a30f61f/2024-02-05--19-41-03`
- `2bbe12792a30f61f/2024-02-05--19-29-55`
- `2bbe12792a30f61f/2024-02-05--18-23-35`
- `2bbe12792a30f61f/2024-02-05--18-22-28`
- `2bbe12792a30f61f/2024-02-05--18-19-28`
- `2bbe12792a30f61f/2024-02-05--18-17-33`

### Provide a route where the issue occurs

2bbe12792a30f61f|2024-02-05--19-29-55

### openpilot version

0.9.6

coffee-cake-isaac, 2024-02-06:
Echoing this as I just checked one of my routes from a newly installed master

112e4d6e0cad05e1|2024-02-05--20-57-37--0
112e4d6e0cad05e1/2024-02-05--20-55-58
112e4d6e0cad05e1/2024-02-05--20-56-29 (Interestingly, here's a route that didn't experience it)

adeebshihadeh, 2024-02-06:
It's from this https://github.com/commaai/openpilot/pull/31089/files#diff-717c4ee865e639b80d8a67bfeaa6dcdbf8b9604b5a3550b3a4d038f5a67a0bb1R176. I'll get this fixed soon.

PR #31391 (2024-02-12): SubMaster: improve service aliveness checks
Fixes #31311, plus adds new checks for services that aren't polled on and checks for too high frequency
closed 2024-02-12 by adeebshihadeh