commaai/openpilot #31336: KIA K5 2024 HEV  fingerprint

lssqla, 2024-02-06:
### Describe the bug

KIA K5 2024 HEV 
dongle id 065323ce3e83f24b 

{'carParams': {'alternativeExperience': 1,
'autoResumeSng': True,
'carFingerprint': 'mock',
'carFw': [{'address': 1988,
'brand': 'hyundai',
'bus': 0,
'ecu': 'fwdCamera',
'fwVersion': b'\xf1\x00DL3HMFC AT KOR LHD 1.00 1.02 99210-L2500 230911',
'logging': False,
'obdMultiplexing': True,
'request': [b'"\xf1\x00'],
'responseAddress': 1996,
'subAddress': 0},
{'address': 2000,
'brand': 'hyundai',
'bus': 0,
'ecu': 'fwdRadar',
'fwVersion': b'\xf1\x00DL3_ RDR ----- 1.00 1.01 99110-L2500 ',
'logging': False,
'obdMultiplexing': True,
'request': [b'"\xf1\x00'],
'responseAddress': 2008,
'subAddress': 0}],
'carName': 'mock',
'carVin': '00000000000000000',
'centerToFront': 1.350000023841858,
'communityFeatureDEPRECATED': False,
'dashcamOnly': False,
'directAccelControlDEPRECATED': False,
'enableApgsDEPRECATED': False,
'enableBsm': False,
'enableCameraDEPRECATED': False,
'enableDsu': False,
'enableGasInterceptor': False,
'experimentalLongitudinalAvailable': False,
'fingerprintSource': 'can',
'flags': 0,
'fuzzyFingerprint': False,
'hasStockCameraDEPRECATED': False,
'isPandaBlackDEPRECATED': False,
'lateralTuning': {'pid': {'kf': 0.0}},
'longitudinalActuatorDelayLowerBound': 0.15000000596046448,
'longitudinalActuatorDelayUpperBound': 0.15000000596046448,
'longitudinalTuning': {'deadzoneBP': [0.0], 'deadzoneV': [0.0], 'kf': 1.0, 'kiBP': [0.0], 'kiV': [1.0], 'kpBP': [0.0], 'kpV': [1.0]},
'mass': 1836.0,
'maxLateralAccel': 10.0,
'maxSteeringAngleDegDEPRECATED': 0.0,
'minEnableSpeed': -1.0,
'minSpeedCanDEPRECATED': 0.0,
'minSteerSpeed': 0.0,
'networkLocation': 'fwdCamera',
'notCar': False,
'openpilotLongitudinalControl': False,
'passive': False,
'pcmCruise': True,
'radarTimeStep': 0.05000000074505806,
'radarUnavailable': False,
'rotationalInertia': 3139.534912109375,
'safetyConfigs': [{'safetyModel': 'noOutput', 'safetyParam': 0, 'safetyParam2DEPRECATED': 0, 'safetyParamDEPRECATED': 0}],
'safetyModelDEPRECATED': 'silent',
'safetyModelPassiveDEPRECATED': 'silent',
'safetyParamDEPRECATED': 0,
'startAccel': 0.0,
'startingAccelRateDEPRECATED': 0.0,
'startingState': False,
'steerActuatorDelay': 0.0,
'steerControlType': 'torque',
'steerLimitAlert': False,
'steerLimitTimer': 1.0,
'steerRateCostDEPRECATED': 0.0,
'steerRatio': 13.0,
'steerRatioRear': 0.0,
'stopAccel': -2.0,
'stoppingControl': True,
'stoppingDecelRate': 0.800000011920929,
'tireStiffnessFactor': 1.0,
'tireStiffnessFront': 201087.203125,
'tireStiffnessRear': 317877.90625,
'transmissionType': 'unknown',
'vEgoStarting': 0.5,
'vEgoStopping': 0.5,
'wheelSpeedFactor': 1.0,
'wheelbase': 2.700000047683716},
'logMonoTime': 32801700298,
'valid': True}

### Provide a route where the issue occurs

222

### openpilot version

0.

### Additional info

1

sshane, 2024-02-07:
You already have a PR here, can you re-open that? https://github.com/commaai/openpilot/pull/30910

sydid, 2024-02-07:
pull Request has been closed  
i dont no

sshane, 2024-02-07:
I think you force pushed the branch accidentally, you can re-open a new one if you don't know how to fix.

sshane, 2024-02-07:
@sunnyhaibin if you want to make a PR for this, feel free!

sshane, 2024-02-08:
closing for https://github.com/commaai/openpilot/pull/31349

closed 2024-02-08 by sshane