commaai/openpilot #31356: Enhanced Lane Keeping Options with OpenPilot Disengagement

vijankush, 2024-02-07:
### When OP is not engaged, having an option that **nudges** the vehicle back into the lane when wheels cross over the lane markings.

Currently, OP disables the stock Lane Keeping Assist (LKA) entirely when it's not engaged. This can be limiting for drivers who may prefer a more subtle form of assistance, rather than the all-or-nothing approach of either fully engaged OP or completely manual control.

While I appreciate the clear distinction between OP's active and inactive states, the absence of stock LKA during disengagement does remove a layer of safety and convenience. 

On the other hand, the Always on Lateral (AOL) feature available in forks is a good option but might be too intrusive for drivers seeking a casual driving experience like myself. 

To address this, I propose two potential solutions:

A. Activate OP only when cruise control is pressed ON. This approach would preserve the vehicle's stock LKA/LTA functionality when OP is off, offering drivers the original safety features of their vehicle alongside OP's advanced capabilities (I know this can be complicated to implement)
B. **Preferred: Instead of letting stock LKAS nudging the car in lane (which seems to suck on all vehicles except Tesla), OP can have an option to enable that. It already gives a lane departure warning on the UI.**

Some folks in the comma discord channels agree with this idea and so I thought it would be nice to bring it up for discussion.

adeebshihadeh, 2024-02-08:
This isn't a bug - feature requests can be discussed in the community discord.

closed 2024-02-08 by adeebshihadeh

vijankush, 2024-02-08:
Thanks @adeebshihadeh - added it in the discord. I got confused with the enhancement label