commaai/openpilot #31406: 2020 Hyundai Palisade - Harness Malfunction Error

scottadair, 2024-02-11:
### Describe the bug

About a week or so ago, something was committed to master which broke support for my car. I'm not getting a "Harness Malfunction" error. When I revert to release3, the error goes away.

Trying to figure out which commit broke. I am not using a Comma branded harness, mine is a custom harness that was made before the official ones were released. Has been working fine for many years now. Not sure what's changed in the code in the past couple weeks. 

### Which car does this affect?

2020 Hyundai Palisade

### Provide a route where the issue occurs

4b8ae689eaa74602|2024-02-11--16-36-27--0

### openpilot version

master

scottadair, 2024-02-13:
I reverted this change, and the harness error went away. 

https://github.com/commaai/openpilot/commit/bcd29a2b8af7c7364040716926d5c18ccd89ca7c

Any thoughts @sshane ?

sshane, 2024-02-14:
What is your Discord? We can quickly determine the issue if you're up for trying a branch

scottadair, 2024-02-14:
Yeah can certainly try master and will disable alpha long support.

My discord account is scottieadair

scottadair, 2024-02-14:
On master with alpha long disable, everything seems to work. No dash errors

sshane, 2024-02-14:
Caused by a refactor 2 months ago not checking if it's a CAN FD car + the PR above to get the ADAS ECU: https://github.com/commaai/openpilot/pull/29733

closed 2024-02-14 by sshane

Gr3gK1, 2024-03-11:
I installed the latest release-c3 2 days ago and received the "Harness Malfunction" error yesterday morning. Thought maybe it was a connection issue. Checked everything. Ask connections are perfect. Then the comma worked for a 3-hour drive, but today the same error occured. The conditions were the same - stopping behind a car with comma saying that control is temporarily unavailable. And then all hell breaks loose. The car starts flashing that all assist systems are malfunctioning and comma complains about the harness error. I have a 2021 Hyundai Palisade. Turning the car off and back on doesn't reset the system, but turning off, opening and closing the driver's door, then turning on again brings comma back to the functional state.

PR #31443 (2024-02-14): Revert "Hyundai CAN-FD: Panda safety config assignments cleanup (#29733)"
We can re-do this a bit better I think. Fixes #31406

This reverts commit 8009b11516fdedd964716a414dcefc996dea807a.