commaai/openpilot #31420: [100$ bounty] fix docker caching on submodules

jnewb1, 2024-02-12:
this bounty is to fix docker caching on the core openpilot submodules

https://github.com/commaai/cereal
https://github.com/commaai/opendbc
https://github.com/commaai/panda
https://github.com/commaai/rednose

cached docker image should be pushed to the ghcr.io repo, similar to how openpilot does it. currently this takes 3-4 minutes for every job on the submodules

bongbui321, 2024-02-23:
I believe docker cache in panda is already enabled

jnewb1, 2024-02-27:
> I believe docker cache in panda is already enabled

nice, so we just need the others then?

bongbui321, 2024-02-27:
@jnewb1 Yeah I submitted PRs in all of them waiting for rednose to merge and bounty solved

jnewb1, 2024-02-27:
just merged the rednose one, think this is solved now! thanks!

closed 2024-02-27 by jnewb1

jnewb1, 2024-02-27:
hmm, just made a PR in cereal and it doesn't seem to have cached... https://github.com/commaai/cereal/pull/593

bongbui321, 2024-02-27:
I wonder why it can pull the cache here https://github.com/bongbui321/cereal/actions/runs/8015514916/job/21895802617. I'll look into it

jnewb1, 2024-02-28:
[image]
seems like the package is there?

jnewb1, 2024-02-28:
caching seems to work locally too...
[image]

bongbui321, 2024-02-28:
How about permission of which repo can use the package? though I'm sure `commaai/cereal` has `Admin` permission already since it is the one that pushes the image. 
Earlier I deleted a new image and push a new one, and it can pull docker cache on my cereal repo. if possible can you try pushing a new image and see if it works?