commaai/openpilot #31437: RAM 2020

iRonsideLeys, 2024-02-13:
### Describe the bug

Hey Guys,

I should have posted this a long time ago.. But my 2020 Ram  my head unit (3comm) will lock up sometimes when I start the truck up I don't get it.  At first i thought it was command start issue but it does happen even when I start the truck manually. (Not all the time)  its to do with the lane assist when ever that disables on its own. I then go to the head unit (3comm) and I tried to get it to respond and no go... Nothing just locked or frozen in time where it last was thinking... My fix is unplug and plug back in and then I can go on my trip and it works fine... just on startup its just annoying unplugging from the back all the time... Thanks I appreciate any help on this!!! Casey

### Which car does this affect?

2020 Ram

### Provide a route where the issue occurs

7de6bfe2692741aa

### openpilot version

9.5

### Additional info

My head unit (3comm) will lock up sometimes when I start the truck up I don't get it.  At first i thought it was command start issue but it does happen even when I start the truck manually. (Not all the time)  its to do with the lane assist when ever that disables on its own. I then go to the head unit (3comm) and I tried to get it to respond and no go... Nothing just locked or frozen in time where it last was thinking... My fix is unplug and plug back in and then I can go on my trip and it works fine... just on startup its just annoying unplugging from the back all the time... Thanks I appreciate any help on this!!! Casey

adeebshihadeh, 2024-02-13:
This is a third party head unit?

sshane, 2024-02-14:
if by 3comm you mean comma 3X, can you get and more recent route? There's only one in the last 2 weeks where you're engaging.

iRonsideLeys, 2024-02-14:
It actually happened on the one trip last week we haven't used the truck in
months until last Friday/Saturday. Does that help

Casey

iRonsideLeys, 2024-02-14:
Yes sorry I meant the 3x I had just purchased it back in Nov/Dec of 2023...
Just have not had time to open a ticket

Caeey

sshane, 2024-02-15:
What do you mean by "lock up," and does this only happen on startup, or during driving as well?

Can you put your device on WiFi so we can upload logs of your most recent route?

iRonsideLeys, 2024-02-15:
Locked up meaning screen is unresponsive to hand gesters.  I will get it
connected to wifi tomorrow.

Thanks for your help

iRonsideLeys, 2024-02-15:
Yes also seems to only happen on startup

sshane, 2024-08-20:
@iRonsideLeys does this still occur?

closed 2024-08-21 by adeebshihadeh