commaai/openpilot #31446: " you to check the driving assistance system" message

haram-KONA, 2024-02-14:
### Describe the bug

A warning message asking " you to check the driving assistance system" is displayed and OpenPilot does not work.   I would appreciate it if you could tell me the reason.

### Provide a route where the issue occurs

route id : 1618132d68afc876/2024-02-14--12-13-50

### openpilot version

0.9.5 and 0.9.6 master-ci 

### Additional info

Car : Hyundai Kona Ev 2nd GEN (HDA2)
device : C3X
 Even if I change everything, including the harness box, harness cable, UTP cable, and ODB-C cable, the same error continues to occur.

closed 2024-02-14 by haram-KONA

sshane, 2024-02-14:
Your GitDiff is broken in the logs, but it looks like openpilot is requesting 3 units at a time up, when we set it to 2 for CAN FD. If this is a local diff, please don't open issues that are not relevant to release openpilot

sshane, 2024-02-14:
Did you change anything locally?

haram-KONA, 2024-02-14:
> Did you change anything locally?

There were no hardware changes at all, and I was testing with “SteerMAX” set to 384. Other than that, nothing else was touched.   One additional thing is that USIM used to work well in the past, but recently it did not work, so I used the APN Setting of "KT" in the "Network" menu. If do not set up APN,  can not be able to connect to the Internet.

haram-KONA, 2024-02-14:
> ```
> self.STEER_DELTA_UP = 3
> self.STEER_DELTA_DOWN = 7
> ```
> 
> You are changing what I suspected right here, please do not touch panda safety if you don't understand what is going on here.

Oh, I'm sorry.
I understand what you mean.
I tried to fix it because the warning that the limit was exceeded appeared too often when the car passed a curve.
i will return it to normal condition. Sorry~~~