commaai/openpilot #31460: Compressed_vipc delay in streaming

Marcos64s, 2024-02-14:
### Describe the bug

As compressed vipc is running an accumulated delay is detected on the total latency time, this is due to the network error incrementing as each frame is passed through. I suspect it has something to do with the first square of sensors, but im not sure. Any solutions to this? 
[image]
[image]
[image]
[image]
[image]

### Provide a route where the issue occurs

Not in a route

### openpilot version

0.9.6

adeebshihadeh, 2024-02-19:
I'm not sure what your issue is exactly, and I can't reproduce any delayed frames. Can you post the code or a better description for the things you're plotting?

Marcos64s, 2024-02-19:
The code is exactly the same as it is in the tools/camerastream/compressed_vipc.py (executed as the readme file). This times where gathered by setting the debug to TRUE set on the frame receiving device.

The issue represented in the graph is that the network delay should not increase over time, it should at least be constant or have small variations. Total time should not be an incrementing value, but have a range (based on the stream first measures) of around 100 ms, throughout the entire stream. 

I have a comma3 and the receiving device is a Jetson Orin (idk if this matters) inside the openpilot container, provided on the repo.

adeebshihadeh, 2024-02-21:
I suspect this is related to your setup somehow.

adeebshihadeh, 2024-05-17:
@Marcos64s still an issue?

Marcos64s, 2024-05-24:
Yes, i m testing with new hardware. Soon i will leave an update.

closed 2025-11-13 by Marcos64s