commaai/openpilot #31537: Mazda: Direct sunlight causes OEM camera to disable LKAS

alancyau, 2024-02-22:
### Describe the bug

When driving in direct sunlight, the Mazda factory LKAS system temporarily gets disabled. When this happens, this causes OpenPilot to malfunction. Is it possible to block these messages when the factory LKAS system decides to disable itself due to sunlight?

The posted route should be roughly when this error occurs. Unfortunately, I'm unable to see when the OpenPilot error occurs in Comma Connect UI.

### Which car does this affect?

Mazda CX-5 2022

### Provide a route where the issue occurs

706d8117176562da/2024-02-21--07-59-03/8

### openpilot version

0.9.5-release

sshane, 2024-02-22:
Why is the route in noOutput and blocking all of openpilot's messages in this route?

I see that the `ERR_BIT_1` from CAM_LKAS is sent when this happens, which openpilot checks for some reason, even though we'd be sending LKAS messages and can safely ignore this, like we do on other cars where the camera can fault.

Can the stock adaptive cruise operate when the camera is in this state, however?

alancyau, 2024-02-22:
@sshane I believe all factory assist systems, including the adaptive cruise control, is disabled when the car is in this state. 

I've tried engaging OP when in this state and it won't engage.

Should I put OP in disabled mode and retry to engage the factory adaptive cruise the next time it's in this fault state?

sshane, 2024-02-23:
I'm saying that your route has the panda in a state where it's blocking all of openpilot's messages, despite fingerprinting and with the openpilot enabled toggle on.

alancyau, 2024-02-23:
I'm not sure what was going on with Panda for that route. I didn't attempt to engage OP for that route.

Does the Panda look OK in this route? I engaged OP for a bit.
706d8117176562da/2024-02-20--21-09-59/3

I'll verify ACC the next time I encounter this fault and report back!

sshane, 2024-02-23:
Yep, that one looks fine.

Jafaral, 2024-02-25:
> Why is the route in noOutput and blocking all of openpilot's messages?
> 
> I see that the `ERR_BIT_1` from `CAM_LKAS` is sent when this happens, which openpilot checks for some reason, even though we're sending LKAS messages and can safely ignore this, like we do on other cars where the camera can fault.
> 
> Can the stock adaptive cruise operate when the camera is in this state, however?
> 
> @Jafaral maybe you have some context about using the bit from the camera here?

If I recall, LKAS stops working with the camera is in that state. I believe I tried blocking the error signal and that didn't work, so there could be other signals that need to be blocked too. LKAS and LANEINFO signals has error messages, there could be other ones. LANEINFO is for  dashboard display.

alancyau, 2024-02-28:
While driving today in direct sunlight and with OP engaged, the factory camera faulted and triggered a "red warning". The fault was about pretty short, so I couldn't remember if the car also disabled adaptive cruise. There wasn't enough time for me to put the C3 in dashcam mode and try to enable factory cruise. Not sure if this log helps.

For what it's worth, I'm more cautious driving with OP engaged in direct sun light because of these potential camera faults.

706d8117176562da/2024-02-28--07-54-26/7

Jafaral, 2024-02-28:
Adaptive Cruise will be disabled if OP isn't engaged.

alancyau, 2024-03-05:
The camera faulted from sunlight this morning. With the C3 completely disconnected, I was able to engage the factory adaptive cruise control. Seems like the system just disables LKAS in this state.

Jafaral, 2024-03-05:
On Mazda cars, adaptive cruise control is independent of LKAS (almost), and it has always been that way. You can disconnect the camera, not OP and MRCC works fine. With OP, those features are coupled. I.e, you can't have LKAS without MRCC and vice versa. If OP is engaged they are both active, if you lose LKAS, OP automatically disengages and it disables MRCC. In cases where I lost LKAS in the past and I wanted to MRCC  in the middle of my drive, I simply disconnect the C3.

Back to my "almost" above, the camera is used with MRCC under 19mph. If you lose the camera (camera error), you still get a dashboard error stating that MRCC is not available under 19mph. If you drop below 19mph while having MRCC engaged , it gives you another dashboard error about canceling MRCC.