commaai/openpilot #31571: VW：Skoda Kodiaq RS 2023 EPS low speed turn issue

Saber422, 2024-02-24:
### Describe the bug

The 2023 Kodiaq RS is the performance version of the Kodiaq.
Following the EPS firmware update in 2023, it seems that there are stricter checks on the steering system.
When making low-speed OP (below 6km/h) requested turns, the Kodiaq RS may occur "Steering temporarily unavailable" on OP. Subsequently, the steering assist function locks, LKAS unusable.
To resolve this, the vehicle needs to be turned off and the Electric Power Steering (EPS) needs to be reset before it can be used again on the road.

Notify @jyoung8607 simultaneously, and thank him for his significant contributions to the community.

### Which car does this affect?

Skoda Kodiaq RS 2023-2022 (Europe import to Taiwan version)

### Provide a route where the issue occurs

0bbe367c98fa1538/2024-02-19--06-44-48/0

### openpilot version

0.9.6 master

### Additional info

1.Lock triggered on a left turn at 15:16
0bbe367c98fa1538/2024-02-23--15-15-49/0

2.Lock triggered on a left turn at 16:27
0bbe367c98fa1538/2024-02-23--16-25-45/1

3.Lock triggered on a right turn at 16:35
0bbe367c98fa1538/2024-02-23--16-32-03/3

4.Lock triggered on a right turn at 20:45
0bbe367c98fa1538/2024-02-23--20-41-41/3

5.Steering wheel lock triggered at low speed at 17:48
0bbe367c98fa1538/2024-02-23--17-31-02/16

yy0922218166, 2024-03-22:
I have the same problem.VW tiguan380 R-line 2023

jkbkster, 2024-05-21:
Same issue on 2020 Skoda Superb