commaai/openpilot #31591: 2021 Prius Prime "Openpilot Unavailable" on C3

zvalentine24, 2024-02-25:
### Describe the bug

I continue to get the "openpilot unavailable, waiting for controls to start message" on fully supported 2021 Prius Prime. Don't recall the issue before openpilot 0.9.5 was released. 

I have taken the following steps to troubleshoot:
-Followed various recommendations from discord on similar issues
-Reboot device multiple times each drive (sometimes it will work, others not, and never on first reset)
-Reseat all connections and cables
-Unplug for at least an hour and tried again
-Reflashed the device using AGNOS multiple times
-Replaced RJ45 cable with new
-Replaced right-angle OBD-C cable with new

Sometimes the openpilot will work while restarting the device several times and sometimes will go days without being available despite restarting the device many many times.

### Provide a route where the issue occurs

233ef9dcd57e63d9/2024-02-24--18-43-20

### openpilot version

0.9.5

### Additional info

Dongle ID: 233ef9dcd57e63d9

sshane, 2024-02-26:
I requested upload of a few drives, can you put the device on WiFi?

zvalentine24, 2024-02-27:
I brought the device inside tonight and it is currently connected to the
WiFi, please let me know if you don't get what you're looking for.

LANdrew12, 2024-03-05:
hey @sshane I am also having the same error message on my C3x after I performed the latest 0.9.7 update. The device still calibrated after a reinstall but not before having to restart several times.

My Dongle ID: a4653a9be878a408 

Things I tried already:
I tried reseating cables and still get nothing
Reinstalled OP 
restarted several times

Car:
2020 Volkswagen Jetta SEL Premium

Let me know if you want to try something else.

sshane, 2024-03-30:
@zvalentine24 does this still happen?

zvalentine24, 2024-03-30:
Yes, this still happens every drive. Once I updated to 0.9.6 it normally
doesn’t take as many resets to get the device to recognize the vehicle, but
still clearly not working as it should.

closed 2025-06-13 by adeebshihadeh

zvalentine24, 2025-06-14:
> Unfortunately, I think this is a HW issue with @zvalentine24
> <https://github.com/zvalentine24>'s device. Closing.

That’s an unfortunate analysis since I replaced all the cables initially to
troubleshoot and comma warranty support told me it was software related
back in 2024 when it was still under warranty. Now over a year later out of
warranty we’re saying it’s hardware. Tough luck I suppose. Thanks for
trying to help me out though.

adeebshihadeh, 2025-06-14:
> That’s an unfortunate analysis since I replaced all the cables initially to
> troubleshoot and comma warranty support told me it was software related
> back in 2024 when it was still under warranty. Now [...]

If that's the case, please open a new comma.ai/support ticket and we'll get your device swapped out. Just link to this reply here in your new ticket.

zvalentine24, 2025-06-17:
Thank you. I opened a new hardware ticket #73825 and referenced this
software ticket.