commaai/openpilot #31619: Hyundai: When switching from Long to Stock I am unable to use distance following

BBBmau, 2024-02-27:
### Describe the bug

I had been using stock ACC for quite sometime now, i would use the distance following often where i change the distance of the car in front of me. I had recently changed to long and have been loving it however i still change to stock from time to time and noticed that when changing i am unable to change the distance following on stock like i would before after changing from long.

Setting is hyundai because this might relate to https://github.com/commaai/openpilot/issues/30950 though this could be occurring in order vehicles also.

### Provide a route where the issue occurs

4203853bc02e087a|2024-02-26--16-08-57--205

### openpilot version

765ebc07d52d327a62fe3cc29c42baff082fa46d

### Additional info

on `master-ci` branch

sshane, 2024-02-27:
openpilot longitudinal alpha toggle needs a restart of the car to take effect, did you try that?

closed 2024-02-28 by adeebshihadeh

BBBmau, 2024-02-28:
@sshane yep was aware of this, was more wondering if it would be possible to toggle it without the need to restart the car. After discussing on discord it looks like it's not possible right now with how openpilot is. Although it might be possible to implement on HKG cars