commaai/openpilot #31627: tici: camera feed too dark

royjr, 2024-02-28:
### Describe the bug

Seems to have started around around december, might be software related with users noticing these issues possibly after an update pushed to release.

firstsrif (12/12/2023) - https://discord.com/channels/469524606043160576/954493346250887168/1184142288583143507
[image]

BIGram (12/13/2013) - https://discord.com/channels/469524606043160576/871838269405556736/1184670185232277514
[image]

Simon Kuang (1/13/2024) - https://discord.com/channels/469524606043160576/871838269405556736/1195936661922652170
[image]

mikejakes (2/28/2024) - https://discord.com/channels/469524606043160576/871838269405556736/1212501848699838465
[image]

firstsrif (2/28/2024) - https://discord.com/channels/469524606043160576/871838269405556736/1212534755388690514
[image]

### Provide a route where the issue occurs

master

### openpilot version

master

adeebshihadeh, 2024-02-28:
This only affects displaying the camera feed in some night time scenes. Since the recorded video looks fine, we've decided not to put any effort into at this time.

closed 2024-02-28 by adeebshihadeh

Miniz199922, 2024-02-28:
I have issues in the daytime. The cars are hard to see.