commaai/openpilot #31651: Hidden WiFi network entries not persistent.

FlyMyPG, 2024-02-29:
### Describe the bug

When manually entering a hidden WiFi network, the connection succeeds, but will not automatically reconnect.

1. Enter SSID and password for a hidden WiFi network.
2. Go on a drive until that network connection is lost.
3. Return to where the connection was first established.
4. Observe that WiFi does not automatically reconnect to the hidden network.
5. Observe that repeating the first step still works.

### Provide a route where the issue occurs

N/A

### openpilot version

0.9.6

### Additional info

2021 Nissan Leaf SV (40 KWh) with Comma 3X.

I only got my C3X a month ago (I'm a noob), and my free Connect trial expired last week (no route to share).

MarcoTheDingo, 2024-03-01:
+1 noticed this as well.

adeebshihadeh, 2024-03-07:
Probably a NetworkManager thing. I suspect this is like a 5m fix, should be an easy $100!

PR #31789 (2024-03-10): Add autoconnect support for hidden WiFi networks

Fixes #31651
closed 2024-03-10 by adeebshihadeh