commaai/openpilot #31662: LogReader: asks to use qlogs before checking all sources

sshane, 2024-03-01:
When not specifying a read mode

jnewb1, 2024-03-01:
ah yes this is an issue with auto_interactive + auto_source. auto_source needs to first check rlogs for all sources before then falling back to auto

jnewb1, 2024-03-01:
also, I think if only qlogs exists on some source, auto_source will choose that first even if all the rlogs exist in a later source. Seems like auto_source should always check if rlogs exist first for any soruce

sshane, 2024-03-12:
`5e75f47a17ee2815/00000000--9f9cb10dc2` this says there are some missing rlogs but all are there

gregjhogan, 2024-03-12:
Listing all files for a route or segment prefix should be very fast on any source and knowing all files for all sources up front may make things easier/simpler (instead of checking if individual files exist)

closed 2024-03-13 by jnewb1