commaai/openpilot #31673: test_models: tests are not set up to test cars with multiple pandas in different safety modes

robbederks, 2024-03-01:
`TestCarModelBase` only sets up a single panda safety mode to run unit tests against (the last one in the list). In the Tesla port however, there are two pandas that are running different safety configs (a valid use case of a multipanda setup).

While replaying the test route, this makes the tests fail erroneously, since it cannot send a lateral control message on the longitudinal panda (see CI failure in https://github.com/commaai/openpilot/pull/29947).

We never noticed this in the past, since the previous test route from Tesla was only set up with a single panda (lat only).

robbederks, 2024-03-02:
> We don't really plan on shipping any multi-panda setups (red panda aux is a bit different), so let's close unless you plan on implementing this.

There's two ways around this:
- I can fix up the tests to properly support this. Would probably look quite a bit less clean since you'll likely have to loop over all pandas in all the spots where you do any safety code calls
- I can just get a single panda reference route like for Tesla AP2 without long control

Lmk what you prefer

adeebshihadeh, 2024-03-02:
Let’s do the second option. Any chance the Raven EPS is different and will support blending?

robbederks, 2024-03-04:
> Let’s do the second option. 

Ok will switch to a single panda route.

> Any chance the Raven EPS is different and will support blending?

No, seems like their logic is pretty much identical to my old car (except for the two redundant units now)

closed 2024-03-04 by robbederks