commaai/openpilot #31688: Build Error: tools/cabana (Missing files?)

michaelhonan, 2024-03-04:
### Describe the bug

Building openpilot (master branch) on my comma device results in this:

```
scons: Reading SConscript files ...                                                                                                                                                                                                                                                                                         scons: done reading SConscript files.                                                                                                                                                                                                                                                                                       scons: Building targets ...                                                                                                                                                                                                                                                                                                 arm-none-eabi-gcc -o panda/board/bootstub-panda.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32fx/stm32f4_flash.ld -Ipanda/board -Ipanda -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/bootstub.c                                                                                                                                                                                   arm-none-eabi-gcc -o panda/board/bootstub-panda_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA -DENABLE_SPI -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32h7/stm32h7x5_flash.ld -Ipanda/board -Ipanda -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/bootstub.c                                                                                                                                                                    arm-none-eabi-gcc -o panda/board/jungle/bootstub-panda_jungle.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32fx/stm32f4_flash.ld -Ipanda/board/jungle -Ipanda/board -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/bootstub.c                                                                                                                                                 arm-none-eabi-gcc -o panda/board/jungle/bootstub-panda_jungle_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32h7/stm32h7x5_flash.ld -Ipanda/board/jungle -Ipanda/board -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/bootstub.c                                                                                                                                               arm-none-eabi-gcc -o panda/board/jungle/main-panda_jungle_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32h7/stm32h7x5_flash.ld -Ipanda/board/jungle -Ipanda/board -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/jungle/main.c                                                                                                                                                arm-none-eabi-gcc -o panda/board/jungle/main-panda_jungle.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32fx/stm32f4_flash.ld -Ipanda/board/jungle -Ipanda/board -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/jungle/main.c                                                                                                                                                  arm-none-eabi-gcc -o panda/board/jungle/obj/bootstub.panda_jungle_h7.elf -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32h7/stm32h7x5_flash.ld panda/board/jungle/obj/startup_panda_jungle_h7.o panda/board/jungle/rsa-panda_jungle_h7.o panda/board/jungle/sha-panda_jungle_h7.o panda/board/jungle/bootstub-panda_jungle_h7.o                                                                                               arm-none-eabi-objcopy -O binary panda/board/jungle/obj/bootstub.panda_jungle_h7.elf panda/board/jungle/obj/bootstub.panda_jungle_h7.bin                                                                                                                                                                                     arm-none-eabi-gcc -o panda/board/main-panda.o -c -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32fx/stm32f4_flash.ld -Ipanda/board -Ipanda -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/main.c                                                                                                                                                                                           arm-none-eabi-gcc -o panda/board/jungle/obj/bootstub.panda_jungle.elf -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA_JUNGLE -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32fx/stm32f4_flash.ld panda/board/jungle/obj/startup_panda_jungle.o panda/board/jungle/rsa-panda_jungle.o panda/board/jungle/sha-panda_jungle.o panda/board/jungle/bootstub-panda_jungle.o                                                                                                             arm-none-eabi-objcopy -O binary panda/board/jungle/obj/bootstub.panda_jungle.elf panda/board/jungle/obj/bootstub.panda_jungle.bin                                                                                                                                                                                           arm-none-eabi-gcc -o panda/board/main-panda_h7.o -c -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA -DENABLE_SPI -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32h7/stm32h7x5_flash.ld -Ipanda/board -Ipanda -Ipanda -Ipanda/board -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc panda/board/main.c                                                                                                                                                                            arm-none-eabi-gcc -o panda/board/obj/bootstub.panda.elf -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -std=gnu11 -fmax-errors=1 -Tpanda/board/stm32fx/stm32f4_flash.ld panda/board/obj/startup_panda.o panda/board/rsa-panda.o panda/board/sha-panda.o panda/board/bootstub-panda.o                                                         [... cut ...]
In file included from tools/cabana/cabana.cc:10:
./tools/cabana/streams/socketcanstream.h:5:10: fatal error: 'QtSerialBus/QCanBus' file not found
#include <QtSerialBus/QCanBus>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
scons: *** [tools/cabana/cabana.o] Error 1
scons: building terminated because of errors.
```

### Provide a route where the issue occurs

N/A

### openpilot version

1728355498d1e5307e7e96324b9232692d62eab6

### Additional info

comma three device, dongle id: dc30ef2ad294a732

michaelhonan, 2024-03-05:
Looks like Cabana is being built on the device because `extras` is being included in the build options.

Knocking out the build has allowed OP to build

[image]

adeebshihadeh, 2024-03-06:
This should fix: https://github.com/commaai/openpilot/pull/31736. I only tested replay, and normal on-device builds only build what's necessary for driving.

michaelhonan, 2024-03-06:
> What version of AGNOS are you running? `cat /VERSION`

9.7

michaelhonan, 2024-03-06:
> This should fix: #31736. I only tested replay, and normal on-device builds only build what's necessary for driving.

That fixed it! Cheers.

closed 2024-03-06 by michaelhonan